bitcraze/crazyflie-firmware #48: 32 bytes CRTP packets not supported

ataffanel, 2015-04-13:
The Crazyflie radio supports 32 bytes packets however this line limits the max packet size to 31: https://github.com/bitcraze/crazyflie-firmware/blob/crazyflie2/modules/interface/crtp.h#L33

When the CRTP data size is set to 31 (to send 32 bytes packet), 32bytes packet are dropped somewhere.  This needs to be fixed.

AlexisTM, 2019-02-27:
Isn't it simply because `CRTP_MAX_DATA_SIZE` is the data which is prepended by a byte of the size of the message? It is using `CRTP_MAX_DATA_SIZE+1` for the full message.

https://github.com/bitcraze/crazyflie-firmware/blob/f6637ad54fe55445ce0d574e89a6e275ad52e860/modules/interface/crtp.h#L49-L72

ataffanel, 2019-03-04:
The source of this problem is that CRTP_MAX_DATA_SIZE is set to 30 and not to 31 as it should (which would make 32 bytes max packet including the 1 byte header).

Though last time I tried to fix it (in 2015...), setting CRTP_MAX_DATA_SIZE to 31 was causing 32bytes packet to be dropped somewhere and I never put/had the time to look where the packet was dropped (or if this was still a problem).

AlexisTM, 2019-03-05:
Could it be that the radio adds a CRC for the last byte?

ataffanel, 2019-03-05:
In theory no, the nRF24 radio handles 32 bytes payload independently of the CRC (the CRC is added outside of the payload). Though, for Crazyflie 2.0 we implement the radio ourselves in the nRF51 so it is possible that there is a bug there.

knmcguire, 2020-06-24:
#29 increased the packet size to 60, so that fixes this issue. However we still might need to address the issue that this cannot be even bigger than it already is. But that is not part of this issue!

closed 2020-06-24 by knmcguire