bitcraze/crazyflie-firmware #52: Yaw Rate in Stabilizer Task?

Hoverbear, 2015-06-10:
Hello,
Was doing some code review and noticed this:

```c
      if (rollType == RATE)
      {
        rollRateDesired = eulerRollDesired;
      }
      if (pitchType == RATE)
      {
        pitchRateDesired = eulerPitchDesired;
      }
      if (yawType == RATE)
      {
        yawRateDesired = -eulerYawDesired;
      }
```

[Here](https://github.com/bitcraze/crazyflie-firmware/blob/crazyflie2/modules/src/stabilizer.c#L226-L237). Is the difference in the `yawRateDesired` calculation on purpose?

tobbeanton, 2015-08-11:
Well you found an old mistake. The problem is that the mapping is wrong in many places and inverting the yaw rate here will trigger a lot of changes (cfclient, android, ios, etc). This is a bit troublesome as they all have to be updated at the same time.

ataffanel, 2015-11-16:
This won't be fixed for backward compatibility reason.

closed 2015-11-16 by ataffanel