bitcraze/crazyflie-firmware #75: Crazyflie Startup Startup melody twice

jakestmart, 2015-10-11:
Hi,

I currently have three crazyflies. I have recently just flashed all three the Crazyflie 1.0 using the latest firmware from: https://github.com/bitcraze/crazyflie-firmware

1 out of the 3 starts up fine and only makes the 'start-up' melody once. The 2/3 play the sound 'start-up' melody twice, about 1 second apart.

Where I had:
1. cd directory of frimware
2. make PLATFORM=CF1
3. cload PLATFORM=CF1

It updates the firmware fine, and I am able to connect to the crazyflie via the client.
- Plays start up melody twice
- When I add thrust there is no response

krichardsson, 2015-10-12:
Hi!

Yes, the watchdog has timed out, which in turn resets the processor. When the Crazyflie restarts it detects that it is recovering from a watchdog timeout, fails the self test and prints "SYS: The system resumed after watchdog timeout [WARNING]". The second start-up melody is played when it restarts. 

Which version are you running? Could you please paste the full console log? There is a line at the top (saying something similar to "SYS: Build 10:d5eb851c07b0 (2015.08.1-10) CLEAN") that shows which version you are running.

jakestmart, 2015-10-15:
Hi this is the output from my console log

SYS: ----------------------------
SYS: Crazyflie 1.0 is up and running!
SYS: Build 13:fb8bdbe0ebfe (2015.08.1-13) CLEAN
SYS: I am 0x8725330754526750671FF57 and I have 128KB of flash!
CFGBLK: v0, verification [OK]
IMU: MPU6050 I2C connection [OK].
IMU: HMC5883 I2C connection [OIMU: MS5611 I2C connection [OK].
MPU6050: Self test [OK].
HMC5883L: Self test [OK].
MS5611: Self test [OK].
SYS: The system resumed after watchdog timeout [WARNING]
SYS: No assert information found

krichardsson, 2015-10-15:
Hi!

We currently have some problems related to task priorities, interrupts, usb and i2c that we are working on (#72, #69) and your problems are most likely a duplicate of one of them. These problems became visible when we added the watchdog recently (fast fail for the win!).

What I might find a bit surprising is that you see these problems on 2 out of 3 Crazyflies.

The best workaround for you is probably to build the firmware in DEBUG mode, since that will turn off the watchdog
    make clean
    make PLATFORM=CF1 DEBUG=1

By the way, your problem with the VM is probably because the VM is not configured to use the USB for the crzyradio. See the "configuration" section of https://wiki.bitcraze.io/projects:virtualmachine:index

You may also find useful information and get help on the forum https://forum.bitcraze.io/

Closing this issue as duplicate.

ottodog, 2016-06-13:
I am having this issue still. 

More info from a different forum user:
https://forum.bitcraze.io/viewtopic.php?t=1975

krichardsson, 2016-06-13:
Which firmware version are you using?

ottodog, 2016-06-13:
Any version after 2015.1

krichardsson, 2016-06-14:
Is it still working on one of your Crazyflies but not the other 2? Is it always the same CF(s) that fail or does it work from time to time?

ottodog, 2016-06-14:
I am not OP of this issue.

ataffanel, 2016-06-14:
Have you tried with the latest version from the repos?

ataffanel, 2016-06-14:
So, This seems to actually be the issue #105. I reopened it and lets continue the discussion there.

closed 2016-06-14 by ataffanel