bitcraze/crazyflie-firmware #78: __packed undeclared with arm-none-eabi-gcc 4.7.4

erwincoumans, 2015-10-26:
Using arm-none-eabi-gcc compilation breaks at the use of __packed.
If the __packed is removed all compiles/links fine. Is there any replacement for __packed?

Version of arm-none-eabi-gcc is:
[~/develop/crazyflie-firmware]$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.7.4 20140401 (release) [ARM/embedded-4_7-branch revision 209195]

cstanke, 2015-10-26:
Try the build with 4.9-2014q4. I think that's the required version since the CF1 and CF2 firmware merge.

ataffanel, 2015-10-26:
__packed is defined in cdefs.h file by gcc. Maybe your GCC does not have it.

You can try to change the line https://github.com/bitcraze/crazyflie-firmware/blob/master/hal/interface/usb_conf.h#L258 into:
``` c
  #ifndef __packed
    #define __packed    __attribute__ ((__packed__))
  #endif
```
If it works I can push it.

We actually use either the latest Ubuntu package arm-none-eabi-gcc or gcc from the GCC ARM Embedded from launchpad.

erwincoumans, 2015-10-27:
@ataffanel Yes, that does the job, see pull request #79

ataffanel, 2015-10-27:
It is merged, thanks

closed 2015-10-27 by ataffanel