bitcraze/crazyflie-firmware #81: Downlink on BLE links only sends the first couple of packets and discards everything else

glennvdw, 2015-11-15:
When connecting to the crazyflie 2.0 using BLE (I used the official iOS app as a starting point) not all of the DEBUG_PRINT(..) messages are actually received at the client side.. Only the first 20 or so packets are sent by the crazyflie... Maybe a full buffer?  This only affects BLE,

ataffanel, 2015-11-16:
This looks like an old bug that was since fixed, did you upgrade your Crazyflie with the latest firmware?

glennvdw, 2015-11-16:
The nrf51 was updated using the iOS app. For the stm32 I tried lots of different versions. Currently running the latest commit on this repo.

This person seems to have reported the same problem about a month ago: https://forum.bitcraze.io/viewtopic.php?t=1700

glennvdw, 2015-11-17:
Which version do you recall to be definitely working?

ataffanel, 2015-11-17:
When pushing this change: https://github.com/bitcraze/crazyflie2-nrf-firmware/commit/f19059868f171449a5b18c0254401ef28a07c6b3 the downlink was working (at least I got the full console at startup). I cannot test with the iOS app right now but I am going to implement the downlink in the Android App.

ataffanel, 2015-11-17:
I could reproduce the problem and made a ticket in the NRF firmware (the bug is in the NRF51). I am looking at it.

ataffanel, 2015-11-24:
Fixed in bitcraze/crazyflie2-nrf-firmware#10.

closed 2015-11-24 by ataffanel