bitcraze/crazyflie-firmware #94: Compilation fails in zipped source tree.

cstanke, 2016-02-19:
The addition of the libdw1000 submodule created a new build step (tools/make/check-for-submodules.py) which fails if run from a [zip file of the source tree](https://github.com/bitcraze/crazyflie-firmware/archive/master.zip).

Make fails to compile if you download a zip rather than clone the git repository.

https://forum.bitcraze.io/viewtopic.php?f=5&t=1907

krichardsson, 2016-02-22:
The intention of the check-for-submodules.py script was to notify users that had already cloned the repo, that they have to get the sub modules. Compiling the code for a zip will be a bit tricky, since you will have to get the sub modules as zips separately and copy them into the vendors dir. 

I think there are several things we need to do for this issue:

1. Improve the script check-for-submodules.py to not fail when building a file tree without .git dir.
1. Improve the readme.md to say that zips are not the best way, but to use clone --recursive instead.
1. The long term solution would be to find a way of not compiling all code as a monolith. Building code for all decks into the firmware has not been a big problem so far but now it's starting to itch. One simple solution would be to use a build time configuration where you can declare which decks to build. The drawback would be that users that buy a deck will have to build their own FW to use the deck. A middle way could possibly be to configure default support for the most popular decks, but not for development HW such as the positioning system. Any other ideas?

ataffanel, 2016-02-22:
A note on 3.: Choosing what to build at build time and building only a subset of the decks by default would indeed be easy but a bit sad because we have more than enough flash and ram in the Crazyflie for the foreseeable future to supports all the published deck out of the box. Though if compiling from ZIP is very important we can always compile in 'degraded' mode (a bit like the nRF51 firmware that can be compiled without bluetooth to avoid the Nordic blob).

cstanke, 2016-02-25:
I like the ideas of #1 and #3. I think there are valid reasons for continuing to support building source downloaded in zips (automation perhaps, reluctance to get involved with git, etc...). As far as #1 goes, maybe there is a way to detect missing submodules and either offer to download those zips or simply to flag the user that they're missing. I'm more inclined to think downloading the zips is a good idea (or at least offering to). In that case, it could work much like Arnaud's nRF51 script that pulls the Nordic BLE stuff.

Anyway, my two cents, but I'm just a single voice. :)

krichardsson, 2016-03-09:
My thinking is that git is more or less a required tool to work with Bitcraze projects and that this should be the default "use case". It would be nice to have the possibility to build from a zip though as an option. 

My suggested solution is to write a python script that reads the .gitmodules file and pulls down the zips if the submodules are missing. This functionality is probably easy to add to the check-for-submodules.py script. I close this bug and open a new issue that describes this feature #95.

closed 2016-03-09 by krichardsson