bitcraze/crazyflie-firmware #137: Current firmware unstable on CF2

tompe17, 2016-07-27:
I wanted to look at the Kalman filter functionality so I installed the lates version (compiling it on Ubuntu 16.04).  It worked OK for CF1 in cfclient. But for my CF2 when I connect to it I sometimes get the sensor values and sometimes not.

So is this well know? Or did I mess up building it?

ataffanel, 2016-07-27:
If this  happens mostly when you activate the kalman filter this is a known bug. This seems to happen when the CPU is heavily loaded and the kalman filter is heavy and added to the connection process it sometime triggers the watchdog timer.

We think this is coming from the I2C driver that behaves badly and lock when the system is under heavy load. There is a work in progress to rewrite the I2C driver which should fix this.

tompe17, 2016-07-27:
Well, I just cloned the firmware repa and compiled with make.  Shouldn't the Kalman filter be disable then?   I read somewhere that disabled was the default.

ataffanel, 2016-07-28:
Interesting, how often would you say it is happening? Like 50% of the time or less? And do you know which version you where running before?

I have seen this every now and then, it would be interesting to find a point when this bug got worst, it could give more clue on how to solve it.

tompe17, 2016-07-28:
I installed 16.02 (release version) and that worked OK.

How can I check out that version? Is it a branch?

I would like to try to enable rate control and need a good firmware version to start with.

tompe17, 2016-07-28:
I just tested with our second CF2.  It worked connecting and disconnect for a while.  While a did that it seemed the firmware rebooted (I heard the beeping sound) and not the red light flashes for a second regularly. Not the normal on for a second.

tompe17, 2016-07-28:
Now after power cycling it does not work at all. USB cable connected. Turn it on. The red light blinks in the correct way. Tries to connect. And the red light stops blinking and is off all the time.

But the next time I tested it it worked...

What worried me is if the connect can drop during flight...

tompe17, 2016-07-28:
And now I cannot provoke it by connecting/disconnecting.  Feels like an uninitialized variable maybe?
I saw a watchdog timer in the console:
```
SYS: Crazyflie 2.0 is up and running!
SYS: Build 131:bfd4840d678e (2016.02-131) MODIFIED
002D and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
IMU: MPU9250 I2C connection [OK].
IMU: AK8963 I2C connection [OK].
IMU: LPS25H I2C connection [OK].
DECK_INFO: Found 0 deck memory.
DECK_CORE: 0 deck enumerated
EEPROM: I2C connection [OK].
AK8963: Self test [OK].
SYS: The system resumed after watchdog timeout [WARNING]
SYS: No assert information found
CFGBLK: v1, verification [OK]
IMU: MPU9250 I2C connection [OK].
IMU: AK8963 I2C connection [OK].
IMU: LPS25H I2C connection [OK].
DECK_DRIVERS: Found 6 drivers
DECK_INFO: Found 0 deck memory.
DECK_CORE: 0 deck enumerated
EEPROM: I2C connection [OK].
AK8963: Self test [OK].
SYS: Free heap: 12656 bytes
```

krichardsson, 2016-09-08:
This is most likely related to an issue that has been fixed in the python client lib.

closed 2016-09-08 by krichardsson