bitcraze/crazyflie-firmware #138: Gyro log missing

tompe17, 2016-07-31:
It seems like the gyro logging is not available anymore.

Is it available under some other name?

We are using that information so it would be good to have it available.

krichardsson, 2016-08-01:
Looks as it was removed from the stabilizer task (stabilizer.c) in commit  fc88c6a7b7a1ccea8399df155b6132a0f474147c and moved into the new sensors task (sensors_task.c). The problem is that that when using the sensors_stock.c (default option) the logging is missing. 

@mikehamer could we move it back to stabilizer.c to have the same logging regardless of sensor implementation, or was there a good reason for moving it?

mikehamer, 2016-08-01:
It could be moved back.

The reason was mainly for cleanliness, such that everything sensor related was handled in the sensors file. Another reason was that I wasn't sure how the Kalman filter would handle the sensor data (since the calls to gather sensor data have been removed from the stabilizer when the Kalman filter is in use), and whether the stabilizer would even have access to the raw sensor data for logging. I must have forgotten about this change when merging everything back with the main branch and adding switches for task vs stock compilation. Sorry for the oversight!

krichardsson, 2016-08-03:
I see, thanks @mikehamer!
It also looks as the current architecture does not fully support the Kalman filter implementation, so I think we will have to take a second look at that in the near future, The logging of the sensors should probably be part of that as well. 
I'll move them back to stabilizer.c for the time being.

commit bbcc449b (2016-08-03): Moved logging of sensors back to stabiliser.c. Closes #138
closed 2016-08-03 by krichardsson