bitcraze/crazyflie-firmware #146: Prevent multiple PARAM_GROUP_START definitions of same name

tobbeanton, 2016-09-15:
Is there a way to prevent PARAM_GROUP_START with the same name using compiler preprocessor functionality?

ledvinap, 2016-09-15:
Preprocessor can't help, duplicates can be in multiple compilation units.
But it is possible to remove `static` from
https://github.com/bitcraze/crazyflie-firmware/blob/b63c552355522389aa8e665bd2c83d1668cf4ca7/src/modules/interface/param.h#L85. That way duplicate groups will cause error in linker (duplicate initialized variable).

knmcguire, 2020-06-24:
An update here, about an half a year ago we tried to do this by  reading out the bin-file and see if the same logging variable occurs twice... but this is not as easy as it seems! There is some primarily code here but we abandoned it due to lack of interest. But maybe we can combine it with the param name length check as described here? #588

closed 2021-05-04 by ataffanel