bitcraze/crazyflie-firmware #152: Failing LPS25H temperature self reading

CarlosRDomin, 2016-09-27:
[I see there was a commit 6 days ago (da09442) which is supposed to fix this issue, however it still happens to me].
The issue is the following: sometimes, when I power up my CrazyFlies, some of them start blinking the red led (the one on M1's arm) 4 times in a row, which means an error occurred. When I connect to them using the cfclient and read the console output, last line says:

> LPS25H: Self test temperature [FAIL]. low: -20.0, high: 80.0, measured: -22.49

Some other times, surprisingly, I get the following:

> LPS25H: Self test temperature [FAIL]. low: -20.0, high: 80.0, measured: 89.90

Any idea of what could be the cause? Temperature in my lab is usually around 20-30ºC. Is it a problem with the moving average filter (maybe an initialization issue)? Is the LPS25H_LSB_PER_CELSIUS constant off?

Please let me know if there's any further testing/debugging that can help you guys out :)

tobbeanton, 2016-09-28:
Are you sure you running the latest code? This is typically what happened, getting random temperature data because the temp data was read from the pressure register instead of the temp register (pressure register wraps around when FIFO is activated). Have you tried with code before the sensor reading change? Might be a sensor issue (but most likely still a bug)

CarlosRDomin, 2016-10-04:
My bad, re-flashed the firmware (thought I had after pulling the latest commit, but I guess I hadn't) and problem is solved, temperature is now in the 20-35ºC range as expected :)

closed 2016-10-04 by CarlosRDomin