bitcraze/crazyflie-firmware #160: Is it possible to compile Crazyflie-firmware to x86 architecture binary?

pfsun, 2016-11-10:
Hey,

I know when compiling crazyfile-firmware, it will need ARM toolchain. Also I know there are libraries which are Cortex-M processor related. I was still wondering whether it is possible to compile crazyflie-firmware to x86 arch binary?

Thanks.

ataffanel, 2016-11-11:
Hi,

The Crazyflie firmware is written in C and uses freertos and since FreeRTOS has windows and Linux support is should be possible to compile and run on x86 as long as you re-write the hardware specific driver.

We have tried as much as possible to put hardware related code in the hal and driver folders. You can already see that we have different driver versions for Crazyflie 1 and Crazyflie 2, you would need to re-make them for your platform.

You would also need to look at the linker script since there is some magic happening there with the log and param subsystem, some section have to be present in memory.

I am curious for what reason you would like to compile the firmware for x86?

/Arnaud

pfsun, 2016-11-14:
Thanks Arnaud for your detail answer. I will try it. I work on one dynamic binary analysis solution for drone. Now we do have drone for crazyflie. But my current initial solution only works on X86 architecture. So I just want to have one try for x86 version. Later I will work on ARM architecture.

ataffanel, 2016-11-15:
That sounds interesting, would be nice to be kept updated if there is progress on compiling Crazyflie firmware on x86!

ataffanel, 2018-08-23:
This has now been done! See https://forum.bitcraze.io/viewtopic.php?f=2&t=2929&hilit=sitl&start=10#p14955.

closed 2018-08-23 by ataffanel