bitcraze/crazyflie-firmware #168: Emergency Shutoff

mikehamer, 2016-11-25:
This is to start a discussion about emergency shutoff.

My idea would be turning off the propellors if any of the following conditions are met (perhaps user-configurable in config.mk)

- Body angle too large
- State exceeds some bounds (eg. velocity abnormally high, position in outer space, etc)
- Measured acceleration in x or y exceeds a threshold while flying (since only acceleration in z should be measured, a large deceleration in x or y probably indicates an impact)

Thoughts @ataffanel, @stephanbro?

stephanbro, 2016-11-28:
I like the idea. Seems like something that could be added to the SitAw module.

krichardsson, 2017-01-13:
Sounds like a good idea. 
I guess it also would protect from take-offs before the filter has converged and position has been established. 
When we added python examples for autonomous flight to the bitcraze/crazyflie-lib-python repo we use the logging system to track the variation in the kalman filter to find out if the filter has converged or not. Maybe this functionality should be moved to the firmware instead as part of this issue?

ataffanel, 2017-03-06:
Can we add to the list of condition:
 - Received Emergency shutoff request from the radio

We need a way to send emergency shutoff to the Crazyflie and we could make a module responsible of emergency shutoff that can be set and reset using the radio.

jonasdn, 2021-03-18:
This is implemented as far as I can tell. We have emergency shutoff request from radio, and we have emergencystop in the supervisor module when a tumbled situation is detected in flight.

closed 2021-03-18 by jonasdn