bitcraze/crazyflie-firmware #170: Altitude hold not working

ataffanel, 2016-11-28:
In the current master branch, altitude hold seems to be broken. This seems to come from the work on the position controller (see #136) that broke the control modes contained in the setpoint structure.

stephanbro, 2016-11-28:
This has definitely been broken as part of the position controller changes, as you suspect. I have a fix I will try to push today.

ataffanel, 2016-11-28:
I was about to look at it but if you already have a fix that is great! thanks.

ataffanel, 2016-11-28:
With PR #172 the altitude hold now holds a constant vertical velocity. The current altitude estimator is not outputting vertical velocity which would explain this phenomena.

ataffanel, 2016-11-28:
39831c3b fixes altitude-hold. I also tried to tune a bit the Z estimator (there was an oscillation between the thrust and the height measurement, fixed by filtering the pressure sensor a bit harder).

closed 2016-11-28 by ataffanel