bitcraze/crazyflie-firmware #186: Consider adding doxygen documentation and automated build

ataffanel, 2017-01-03:
In order to have an updated code documentation we could document the Firmware with Doxygen (or other similar program?). The best cuuld be to automatically generate and distribute the documentation for the master branch for every commit. This ticket aims at starting the discussion, there is a couple of things to define:
 - Do we want code-generated documentation?
 - What system to use, is Doxygen still the one to go for?
 - Do we generate the doc automatically for every commit?
 - Do we fail a commit that introduces non-documented exported functions?
 - Any other consideration?

This is related to #148.

krichardsson, 2017-01-04:
>>> Do we generate the doc automatically for every commit?
Yes, if possible. We want to fail fast.

>>> Do we fail a commit that introduces non-documented exported functions?
Yes, I think we should enforce it.

ataffanel, 2017-01-09:
What about using github pages to publish the doc? This way the build script in travis could automatically push the updated doc in a git repos and github pages would make that available automatically.

knmcguire, 2020-06-24:
we started this up here #447, where we created a doc folder in the github repo, and the bitcraze website compiles and publishes the doc. We thought about githubpages but eventually ended with wanted to use our website instead with hosting this documentation since we have  more control on where the documentation is displayed and that we are not depended on other thirdparty solutions hosting our documentation. 

So therefore we can close this issue :D

closed 2020-06-24 by knmcguire