bitcraze/crazyflie-firmware #202: Update VL53 driver for production OW memory content

ataffanel, 2017-02-23:
The final production OW memory content has been decided for the VL53 board, now called Z-Ranger: https://wiki.bitcraze.io/projects:crazyflie2:expansionboards:index#deck_info. The driver needs to be updated accordingly. 

@stephanbro You are using the VL53 driver for ranging in another direction. What about setting the memory VID/PID to 0xBC:0x09 but with a different name (called ID in the wiki). This way the driver will be initialized, test what board is attached, and can take decision as if the values needs to be pushed in the EKF or not (for example).

stephanbro, 2017-02-23:
Since the application I'm using the VL53 for has a fairly custom deck different from the Z-Ranger you guys are working on, I don't have any issues manually setting the VID/PID on my board now that the ranging application has varied pretty decently. Thanks for the consideration though!

Also unless any objects, I don't have any issue with you guys modifying the code to make the Z-Ranger the normal operation of that deck (instead of forward ranging).

ataffanel, 2017-03-01:
Ok, we keep it as it is then. Thanks.

closed 2017-03-01 by ataffanel