bitcraze/crazyflie-firmware #204: Update setpoint structure to handle new controllers

ataffanel, 2017-03-06:
In order to support new controller like Quaternion non-linear controller from @mikehamer and Se(3) controllers from @mgreiff and @whoenig we need to shape-up the setpoint_t structure to add the required fields.

Emergency stop is planned to be handled independently of the setpoint.

This is a proposal:
``` c
#define SP_POSITION 0
#define SP_VELOCITY 1
#define SP_ACCELERATION 2
#define SP_JERK 3

#define CTRL_ENABLE_POSITION (1<<SP_POSITION)
#define CTRL_ENABLE_VELOCITY (1<<SP_VELOCITY)
#define CTRL_ENABLE_ACCELERATION (1<<SP_ACCELERATION)
#define CTRL_ENABLE_JERK (1<<SP_JERK)

#define CTRL_ENABLE_ABSOLUTE_ANGLE 0
#define CRTL_ENABLE_RATE_ANGLE 1

typedef struct setpoint_s {
  uint32_t timestamp;  // Timestamp in "tick" of the stabilizer loop

  // Position setpoint in the global frame. Indexed with SP_...
  float x[4];
  float y[4];
  float z[4];

  // Bitfields containing CTRL_ENABLE_...
  uint32_t modeX;
  uint32_t modeY;
  uint32_t modeZ;

  // Angle setoint. Will be ovewritten by the position controller if enabled.
  quaternion_t attitude;   // Attitude setpoint in the global frame
  vector_t attitudeRate;  // wx, wy, wz angular rates around the local bodyframe in rad/s

  // Bitfield containing CTRL_ENABLE_ABSOLUTE_ANGLE and CTRL_ENABLE_RATE_ANGLE
  uint32_t attitudeMode;

  // Thrust, overwritten by the Z position controller if enabled
  float thrust;  // Thrust between 0 and 1.
} setpoint_t;
```

Now the request for comment:
 - How does this look?
 - Do we also need the "SNAP" (derivative of Jerk)?
 - Anything else needed in the setpoint structure?

whoenig, 2017-03-06:
I am not a big fan of those enable-flags. It basically creates a lot of different possibilities, it is impossible to test them all with a single controller and even worse if there is a selection of controllers. Furthermore, it creates weird/implicit dependencies (this controller assumes that those flags are set...)

In general, I think there are two kinds of controllers:

- Full controllers, i.e. you give them a trajectory (position, velocity, acceleration, attitude rate, yaw) and they will compute the required attitude automatically;
- Attitude-controllers, i.e. you give them the desired attitude (angle or rate) and thrust and a pilot does the rest (position related part)

Most things I am aware of can be done using this approach. For example, for a simple waypoint controller, you can set velocity/accelerate/desired attitude rate all to 0 and it will work (at low gains).

I would rather have two kinds of controllers (and two setpoint structures) and thus be able to rely on the compiler to find any use-mismatches.

mikehamer, 2017-03-06:
I think being able to enable/disable different levels of control is extremely useful. The quaternion-based controller I wrote works well on the different levels mentioned, and I am often switching mode, particularly between position and velocity control modes.

However perhaps it would be better to define and use (as I have when I use my quaternion controller)
```
CONTROL_MODE_POSITION = CTRL_ENABLE_POSITION | CTRL_ENABLE_VELOCITY | CTRL_ENABLE_ACCELERATION
CONTROL_MODE_VELOCITY = CTRL_ENABLE_VELOCITY | CTRL_ENABLE_ACCELERATION
etc
```
since this better reflects how the controllers would be used.

I think this level of access is important to keep open, to maximize the utility of the Crazyflie to developers. At the end of the day the end user won't be interacting with this level anyway (unless they want to), so API developers can simply enable the modes that suit their purpose and the end user is none the wiser.

On a different note, as far as I see having jerk in addition to acceleration and body rates could be redundant (since body rates can be calculated from acceleration and jerk, with the exception of yaw rate which is free).

whoenig, 2017-03-07:
I understand the reasoning behind that, but it seems odd to me to put it in the setpoint structure. It seems like those things should be parameters to the individual controller instead? That way, you can just add (and test) the various combinations you really support and care about.

Also, because the Crazyflie is an open platform, the (internal) API is more important than the API of a closed-source product (and the end-user is essentially a developer here). It should be as simple as possible to allow others (who might not have a strong CS background) to contribute as well.

knmcguire, 2020-06-24:
#236 added support to indicate in which stab_mode_t each variable in position or attitude set point should be handled. So it seems as this issue has already been fixed with that update.

closed 2020-06-24 by knmcguire