bitcraze/crazyflie-firmware #209: Uninitialized variable in bmi160_set_power_mode

markhermelinggt, 2017-03-21:
Hello,

For a side project I am doing static analysis on the crazyflie firmware and I found a problem in bmi160_set_power_mode. It defines a local variable data, but never initializes it. It then starts to doing some register calculation with it starting in line 468 with:
temp = data & ~BMI160_ACCEL_UNDERSAMPLING_MASK;

Problem is data is never initialized, so the value of temp is undefined in this.

Not sure how to fix this easily. Not sure if it best to just change this to 
temp = ~BMI160_ACCEL_UNDERSAMPLING_MASK;

Thanks,
Mark

tobbeanton, 2017-03-30:
Thanks for the reporting. If I read the code correctly the data variable is [read](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/drivers/src/bmi160.c#L465) from the register a bit earlier so only uninitialized if the read fails. One could then argue that the write later will also fail and the problem will be detected, but the right way would be to check the read as well. Or am I missing something?

markhermelinggt, 2017-03-30:
Yes, your description is correct. It all depends on how robust you want to be. It could be that the read fails and the write succeeds, in which case you have a failure case that would be blissfully hard to debug. 

I think either checking the read, or providing an valid initial value would be a good solution.

The question is: How robust do you want to be in the firmware, there are other places where this pattern occurs, I am happy to provide a list if you think that will help. They all follow the same pattern: Variable may not be initialized properly if the failure path through the get or read function happens.

tobbeanton, 2017-04-10:
The firmware is probably full of dangers and if the list is not to long it would be great to have.

This code is written by Bosch though so I will try and forward it to them.

markhermelinggt, 2017-04-10:
Tobias,

Perfect, I’ll craft a list and send it over in a few weeks.

Regards,
Mark

—
Mark Hermeling | GrammaTech | Senior Director Sales and Marketing
mobile +1 (607) 351-5719 | www.grammatech.com

krichardsson, 2019-08-20:
Seems as this file has been updated since this issue was reported and the problem is now solved. Closing.

closed 2019-08-20 by krichardsson