bitcraze/crazyflie-firmware #214: Add profiling framework

krichardsson, 2017-04-04:
Currently we have functionality to get CPU usage information per task. In some cases it could be useful to be able to profile the code in greater detail. Investigate the posibilites and possible solutions.

theseankelly, 2018-04-16:
I did some work to use a high-res tick counter and plumb latency measurements through to the existing logging framework (opt-in at compile time) but I never finished it off. 

Did you have any third party frameworks/tools you hoped to plug in to for a profiling framework, or were you hoping to use the logging framework?

If you want I can spend some time polishing up what i worked on and submit it for PR.

krichardsson, 2018-04-16:
I did not really have any special tools or frameworks in mind, just that profiling can be useful. A benefit of using the existing logging fw is that no external tools are required.

It would be awesome if you want to add it!

theseankelly, 2018-04-17:
Gotcha. A downside of this approach is that simply enabling the logging-based profiling does affect what you're trying to profile. So, depends on the necessary accuracy of the measurements. Probably still useful enough for simple ballpark measurements of the code -- I'll give it some more thought and clean it up.

krichardsson, 2018-04-17:
Agree, I think it would be useful even though it might not be perfectly accurate in all situations.

closed 2021-08-30 by jonasdn