bitcraze/crazyflie-firmware #237: Add parameters to check that deck driver have been enabled

ataffanel, 2017-07-01:
It would be useful for diverse client to be able to check that a deck driver has been enabled or not. 

Today this is done in the PC client checking for the presence of the one-wire memory contained in the decks but being able to check for driver initialization would also allow to check a manually enabled driver.

Doing so with parameters is also easier to use than parsing the one wire memories.

The proposal is to have one read only parameter per driver with the name "deck.<driver name>". The content is a boolean, 1 for initialized, 0 for not initialized.

fredg02, 2017-11-03:
I noticed that driver initialization seems to be independent from the actual existence of a deck. For example, the deck driver for the zRanger is initialized even when the deck is not actually connected. This is in line with the description in this bug, but not expected by the user, I think

Is this working as intended or is this a bug?.

krichardsson, 2017-11-03:
@fredg02 , that is not the intention. Only drivers for decks that are detected or drivers that are forced using the compile flag (like ```CFLAGS += -DDECK_FORCE=bcZRanger``` in config.mk) should be initialized.

Can you describe how to reproduce it?

fredg02, 2017-11-03:
I used the Python client to connect to a CF2 without a zRanger or Flow deck. In the parameters tab the ```deck.bcZRanger``` parameter showed "1", when it should have showed "0".

krichardsson, 2017-11-03:
What firmware are you running?

krichardsson, 2017-11-03:
Yes, that sounds like the z-ranger driver has been initialized, the question is why.
I can not see the same behaviour on my CF with (more or less) the same FW. Is the firmware a clean build without any flags set in config.mk?

fredg02, 2017-11-03:
AFAIK it was a clean build, but I will do some more testing to isolate/reproduce the problem.

fredg02, 2017-11-07:
OK. So apparently I don't know how visibility of global variables in C works. :(
Deck initialization works as expected, but I was testing a new parameter for the buzzer deck and used the ```isInit``` variable, which is also used by the ```zranger.c``` class. That caused the issue I was seeing. Sorry for the noise.

krichardsson, 2017-11-08:
@fredg02 The isInit should have been static to avoid interference with other modules. Fixed in #262

commit efbd2d09 (2018-08-07): Completed read only params to indicate if deck drivers are initialized or not. Closes #237
closed 2018-08-07 by krichardsson