bitcraze/crazyflie-firmware #244: Warning while making

prasgane, 2017-08-05:
Hello,

When I compile it with make, it get the following warning:

`make[1]: Leaving directory '/home/~/Crazyflie/crazyflie-firmware/tools/make/cmsis_dsp'
  LD    cf2.elf
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-atexit.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-ctype_.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-errno.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-exit.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-fini.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-impure.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-init.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-memcpy.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-memset.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-strcpy.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-strlen.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-strncmp.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-strncpy.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-strtol.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-__atexit.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv7e-m/fpu/libc_nano.a(lib_a-__call_atexit.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
`
Can someone please help me fix this.

Thank you

ataffanel, 2017-08-07:
Hi, There seems to be a setting miss-match between your installed lib and the compiled Crazyflie firmware. What GCC are you using and on what platform?

prasgane, 2017-08-07:
Hi,

The current GCC version is 5.4.0.
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0

ataffanel, 2017-08-08:
Are you using the arm-none-eabi from ubuntu or the one documented there: https://github.com/bitcraze/crazyflie-firmware#debianubuntu? I do not think we have tried using the GCC from Ubuntu so it might be the problem.

prasgane, 2017-08-08:
I am using the one documented on the github page.

prasgane, 2017-08-08:
I just removed it and when I installed it again, I got the following errors:

```
W: The repository 'http://ppa.launchpad.net/terry.guo/gcc-arm-embedded/ubuntu xenial Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://ppa.launchpad.net/terry.guo/gcc-arm-embedded/ubuntu/dists/xenial/main/binary-amd64/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

prasgane, 2017-08-09:
A clean install of arm-none-eabi does not solve this problem. Any more suggestion?

krichardsson, 2017-08-09:
Some pointers on how this is installed in a few different environments might help:

1. You can see how it is installed in the docker image that we use to build the firmware on the build server. Take a look at https://github.com/bitcraze/docker-builder/blob/master/src/Dockerfile

1. For the virtual machine set up, see https://github.com/bitcraze/bitcraze-vm/blob/master/createBitcrazeVM.sh

ataffanel, 2017-08-10:
One other pointer: I think I recognize the problem, even after adding the PPA Ubuntu keeps using the package from the ubuntu repos, this I guess is a bug.

On my computer I downloaded the compiler from there: https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads, decompressed it in my home folder and added it to the path by adding at the end of ~/.bashrc: 
``` bash
export PATH=$HOME/opt/gcc-arm-none-eabi/bin:$PATH
```

prasgane, 2017-08-25:
So, I tried what Arnaud suggested first but that didnt fix the problem and also tried what Kristoffer suggested and I ended up breaking my toolchain. I tried downloading and building the firmware on 2 different computers and asked a colleague in my lab to do that same and all of them ended up getting the same warnings.

ataffanel, 2017-08-28:
Just a side note to start: we are not using wchar_t anywhere in the firmware so this warning should be harmless. It needs to be fixed anyway.

This problem really looks like you are not running the right version of gcc. For reference this is what I am running, can you run the same commands to see what you are running and where it is installed (ie. if it is installed by apt-get or by you):
``` bash
$ which arm-none-eabi-gcc
/home/arnaud/opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

prasgane, 2017-09-16:
Hello Arnaud,

Sorry for the delay. I finally got the opportunity to work on this again and like you had guessed, doing sudo-apt-get install arm-none-eabi-gcc install 4.9.3. I went ahead installed the latest update and then changed my .bashrc and the warnings did not show up when compiling! 
I just needed to get 6.3.1 and not 4.9.3. 
Thank you so much!

ataffanel, 2017-09-22:
Hi, great that this is fixed.

closed 2017-09-22 by ataffanel