bitcraze/crazyflie-firmware #259: Freefall detection with Flow deck

fredg02, 2017-11-03:
Freefall detection with Flow deck should activate position hold (at default height, e.g. 40cms) and landing sequence after a little delay (e.g. 5 sec).

krichardsson, 2017-11-15:
It might work, assuming it is free falling with the correct orientation in the air so that the laser ranger can measure the distance to the ground. There is also a risk that the Kalman filter has not converged and we don't understand where we are. It is worth a try though.

knmcguire, 2020-06-24:
Its a nice idea, but I personally think that the Flowdeck is not suitable for this and this should probably be done with the IMU instead or an external positioning system. There is already a sitAw module that detect this so this could be a really cool application. However I don't think this should be actual part of the firmware but separate as with the app-layer. 

As this is an old issue with low activity I will close it for now !

closed 2020-06-24 by knmcguire