bitcraze/crazyflie-firmware #264: Undefined reference to uwbTdoaTagAlgorithm in locodeck.c:161 when compiling with LPS_TDOA_ENABLE=1

benkuper, 2017-11-11:
Hello,
i'm using the BitcrazeVM, i launched the "updateAllProjects" and then tried to compile with config.mk including LPS_TDOA_ENABLE=1 and the compiling fails with that : 
Undefined reference to uwbTdoaTagAlgorithm in locodeck.c:161

when not setting LPS_TDOA_ENABLE=1 , it compiles fine.

krichardsson, 2017-11-13:
Are you sure the update worked as expected? There is not really a reference to uwbTdoaTagAlgorithm on line 161 in locodeck.c in the latest version of the file.

Update from the command line
```git pull```

Maybe try to clean before building as well?
```make clean```

benkuper, 2017-11-13:
Yes really sorry for that i issued too fast, i actually added the lines CFLAGS += [...] instead of directly LPS_TDOA_ENABLE...
Closing the issue as it's not a really problem, but it might be good to add to the config.mk.example these lines commented out :)

closed 2017-11-13 by benkuper