bitcraze/crazyflie-firmware #267: Make CF2 platform fails from freshly cloned master. Spurious loader flag?

JGSuw, 2017-11-17:
Hello,

I have just revisited crazyflie-firmware after about two months. I recloned the repository to make sure my build system was ready, and I ran into an issue with make.

When I attempt to

`$ make PLATFORM=CF2`

I receive the error

`arm-none-eabi-gcc: error: nano.specs: No such file or directory`

Which is caused by this [line](https://github.com/bitcraze/crazyflie-firmware/blob/master/Makefile#L275).

It would appear that either this flag is referencing a file which should be in master but is missing, or this flag was incorrectly merged into the master branch.

krichardsson, 2017-11-17:
nano.specs seems to indicate that your problem is related to your compiler/toolchain.
What OS/build environment are you running? 
If you are using an installed compiler (usually on linux), check that all you need is installed. 
Other solutions cold be to use the toolbelt or VM where everything that is needed should already be installed (see the README.md for details).

By the way, PLATFORM=CF2 is not needed any more. Since we removed support for CF1, CF2 is the default setting.

JGSuw, 2017-11-17:
Thanks for the reply,

I'm building on Ubuntu 14.04 LTS 64b, using Terry Guo's arm embedded toolchain. My build system was set up according to the crazyflie-firmware README.md. Basically I just installed lib-newlib-arm-none-eabi using apt-get. 

I would love to track down what my toolchain problem is. I would prefer not to have to use another virtual machine or fuss with setting up a Docker image. Hopefully I won't have to start from a fresh VM.

krichardsson, 2017-11-20:
Great, I hope you succeed!
Please share if you find out.

Please close if you feel this issue is complete.
Thanks!

closed 2017-11-21 by JGSuw