bitcraze/crazyflie-firmware #268: FlowDeck causes EEPROM communication failure.

JGSuw, 2017-11-17:
Hello,

I recently acquired the optical flow deck, and am eager to use it with my Crazyflie 2.0. Unfortunately, using the latest firmware from the master branch, the system fails to initialize when the flow deck is installed.

The failure occurs in [system.c](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/system.c#L169), and is caused by [this function](https://github.com/bitcraze/crazyflie-firmware/blob/376a1de0dbadf300411c25a04c6f15341259ea43/src/drivers/src/eeprom.c#L61).

I have the flow deck installed in the correct orientation, with the camera facing downward and the forward marker pointing in the same direction as the marking on the bottom silkscreen of the crazyflie.

Has anyone else run into this issue? I do not have this issue when I use one of my older branches of crazyflie-firmware, from before the flow deck was released.

JGSuw, 2017-11-17:
Hello again,

Thanks for the reply. As someone who is constantly hacking on things, I usually build the firmware myself. Fortunately, I was able to get it running using the 2017.06 release, which I built from source.

I think I may have an issue with my toolchain which I need to pin down. However, since I was able to build 2017.06 and run it on the crazyflie successfully, I'm not sure that this particular issue was caused by my toolchain. It could be though, who knows?

Over the summer I was working with a 3rd party expansion board which interfaces with the Crazyflie ARM processor via the I2C bus, and ran into basically the same issue where the EEPROM communication would fail during the system initialization. I wonder if that issue I was having is related to this issue.

ataffanel, 2018-08-23:
I am cleaning-up old issues and since this problem is not observed in the master branch of the firmware I close it. Do not hesitate to re-open the issue with more info about how to reproduce the problem if it persists.

closed 2018-08-23 by ataffanel