bitcraze/crazyflie-firmware #271: CrazyFlie2.0 shuts down when it enters the LPS area

KhristopherK, 2017-12-07:
Hi,
I'm not sure how to check where the error is, but we are having an issue when the cf works well when not using LPS. however, as soon as it enters the LPS area and the first message is received (as seen by the LED on the anchors and the deck) the cf shuts down and can't be restarted (needs battery reconnecting).

Is there a place where the CF logs to when connected to the radio and the python client to check it out? I'm 99% sure it's user error (something I did or didn't do) so would like to check it out.

Thanks! looking forward flying many of these!

krichardsson, 2017-12-07:
Hi!

Are you using Two way ranging (the default mode) or TDoA?
There is a (semi) known problem in the Kalman filter that we have seen that causes the Crazyflie to crash. We don't know what the reason is but it seems to happen from time to time in TDoA mode.

> can't be restarted (needs battery reconnecting).

This sounds a bit surprising and does not really fit with the known problem. What should happen in case of fatal error (divide by zero, try to access unmapped memory or similar) is that the Crazyflie should turn on both front LED with a red light and freeze up. 
If your firmware is compiled with the watchdog active (default) it should restart a few ms later and end up in the "failed init" state with the M1 LED flashing rapidly 5 times in a sequence. In this state you can connect to the Crazyflie with your python client and check the Console tab to see some information of the cause of the restart.
What I find surprising is that it should be possible to restart the Crazyflie using the power button in this state and you should not have to disconnect the battery.

The NRF is handling power management of the STM32 and if you have to disconnect the battery to restart, this might indicate a problem in the NRF instead. Have you modified the NRF code in any way (the https://github.com/bitcraze/crazyflie2-nrf-firmware repo)?

KhristopherK, 2017-12-08:
Hi, thanks for responding. I have not made any mod to the firmware yet. All I've done since receiving the units is update the firmware to the latest found in github. I am using all defaults as I want to test the system first. 

I have since been able to get a few nodes working. So far I can only get 3 to show up now with the vehicle not restarting. I will keep working on this today and let you know if I can get to work properly. I still think is user error, as I am learning as I go. 

Regarding the battery disconnect. I'm still having to do this, I am yet to see the vehicle crashing gracefully. It just goes off and I have to restart it as the button doesn't seem to be doing anything for me. I am able to turn it off and on with the button until it crashes, where the battery pull is needed.

thanks, and I'll keep you posted on this.

KhristopherK, 2017-12-08:
Hi,

Updates, I'll be closing this issue as I can not replicate it anymore. The devices seem to be working and communicating properly. 

Thanks!

closed 2017-12-08 by KhristopherK

krichardsson, 2017-12-09:
Hi!

It sounds as there might be a problem with the power button. Could you
please drop us an email at support@bitcraze.io for further investigation.

br
Kristoffer