bitcraze/crazyflie-firmware #291: Radio address not always set on start up

krichardsson, 2018-02-05:
The radio address of the radio is set at start up, if no setting is found it is set to the default 0xE7E7E7E7E7. It seems as the radio channel is not always set to the programmed address but ends up at the default address.

How to reproduce:
1. program the CF to use a channel (other than 0xE7E7E7E7E7)
2. start the CF
3. verify the address. 

In most cases it works as expected, but the default address has been observed from time to time

knmcguire, 2020-06-26:
With the current master, I tried restarting the crazyflie about 30 times and I was not able to recreate this problem. Also it does not seem that anybody else is experiencing this as well.

knmcguire, 2020-06-26:
Alright, the chance of this happening is very slim, but it might be a timing issue. If we make a script that start and restarts the crazyflie many times, then we might be able to catch this.

closed 2021-08-30 by jonasdn