bitcraze/crazyflie-firmware #292: Position controller take off Z over shoot

krichardsson, 2018-02-05:
The PID position controller sometimes over shoots wildly in Z when taking off. The phenomena is not thoroughly investigated but has been observed with the LPS system in TDoA mode.

How to reproduce:
1. Create a python sequence where the setpoint is moved to 0.4 meters height
2. Run the sequence 

Observations:
* It does not happen every time. Could it be related to something that is not properly reset?
* It is not connected to a specific CF individual
* Might be related to the LPS system. Possibly missing position updates from LPS in combination with noisy Z-accelerometer?

jonasdn, 2021-09-06:
Has anyone seen this lately?

krichardsson, 2021-09-06:
I'm fairly sure it is till there

knmcguire, 2021-09-06:
I can confirm this as well, Seen this with the LH system too. Wondering if it is a control tuning problem or not

jonasdn, 2021-09-08:
This _feels_ like an estimator issue. It would be great to get some more test and feedback on this!

If anyone would be willing to do a script of what @krichardsson describes above and see if they can reproduce this in different scenarios, that would be great!

knmcguire, 2022-02-17:
So me and @tobbeanton have looked at this a bit, and tuning down the z PID controller a bit and reducing the maximum velocity it can fly. The problem mostly is, that from ground to an height is usually quite unstable, and when flying from an height to a higher height is usually quite snappy and even overshootless with the current gains.

This instability at startup seems to happen in two situations:
- Low or bad battery
- Worn out motors

We tried to put the idle motor thrust (powerDist.idleThrust) to a certain value (like 5000 or 10000) and it seems to take off much more stable. This reduces the impact of the sudden motor start to an height which can cause a dip in the battery supply, causing all kinds of issues. 

We need to implement perhaps some kind of ramp-up functionality in the crazyflie firmware between the state of being on the ground and the mode inFlight.

knmcguire, 2022-09-06:
We need to check how we can protect 'misuse' of the commander setpoints, either within the firmware or cflib. Anyway, this particular issue can not be fixed in the controller, just in the way we use it.

closed 2022-09-06 by knmcguire