bitcraze/crazyflie-firmware #301: Adding more anchors for TDOA

AlexisTM, 2018-03-16:
To add more TDOA anchors is it enough to add the anchor configuration for the loco deck or do we need to add something in the anchors? (TDMA?)  https://github.com/AlexisTM/crazyflie-firmware/commit/a53aad31456c8e4496c88fa72b45f3b7de804e55

krichardsson, 2018-03-19:
It will unfortunately require quite some work to use TDOA with more than 8 anchors. The main obstacle is the TDMA between anchors and the anchor code would have to be modified.

The two main tracks we have been discussing are:

1. Modify the number of anchors in the TDMA scheduling. 16 is probably fairly easy to implement but there will be an upper limit on how many anchors that can be supported. If extending the current solution where anchor 0 is the master anchor, all anchors must still be in range to to anchor 0 which limits the physical area that can be covered. Anchor 0 would still be a single point of failure.

1. Let anchors transmit at random times. We have not investigated this closely but it has the potential to remove a lot of the problems we see today. The idea is that anchors transmit their messages at random times and that there is not central scheduling. The UWB radios should be pretty good at handling collisions but this concept requires more investigations.

We currently do not have any plans on working on this but if you are interested in investigating it we would be more than happy to support you with knowledge and ideas :-)

AlexisTM, 2018-03-24:
Thanks a lot for your answer!

Badly, I will not be able to invest time in it right now. For the time being, we can use only 8 anchors.

krichardsson, 2018-06-15:
@AlexisTM You might want to try out the TDoA3 implementation, see https://wiki.bitcraze.io/doc:lps:tdoa3:protocol
It is still a bit experimental and not fully supported in the client but it woks pretty well.

closed 2018-06-15 by krichardsson

AlexisTM, 2018-06-15:
This is for sure what I will do! TDoA3 is some serious feature! Thanks!