bitcraze/crazyflie-firmware #307: the zRangerReadRange adds a hard-dependency between one deck and the core Crazyflie firmware

ataffanel, 2018-04-04:
The function [zRangerReadRange](https://github.com/bitcraze/crazyflie-firmware/blob/e9e84fd92eafb62789c402fb228c1fe30c0d7b6b/src/deck/drivers/src/zranger.c#L119) is called by [sensors_cf2.c](https://github.com/bitcraze/crazyflie-firmware/blob/ae78057001a8ea79879ad60d7f38f688637a2fac/src/hal/src/sensors_cf2.c#L188). This adds a hard-dependance between one deck and the core Crazyflie code which is not good.

For instance, this prevents the implementation of a sensor deck implementing absolute height which would be used instead the z-ranger.

One solution would be to do the same as for the kalman filter: having the z-ranger pushing measurement in a queue. Another would be to have the deck driver register itself as a 'zrange provider'.

As a more general note: We do not have fixed rules when it comes to deck driver dependencies, maybe we should make some. It seems reasonable to not have the 'core' of the crazyflie calling directly into a deck driver.

krichardsson, 2019-10-31:
@ataffanel It seems as the code has moved a bit since this issue was created :-) There are some similar problems in the current code base though.

It seems as zRangerReadRange() is only called from the various sensor implementations, in the XXXAcquire() functions (for instance sensorsBmi088SpiBmp388Acquire()).
The sensorsAcquire() function in turn, is only called from estimatorComplementary() and stabilizerTask(), where stabilizerTask() does not use the zrange data. The only use of the zrange member is in position_estimator_altitude.c, called from the estimatorComplementary().

So it seems as the call to zRangerReadRange() adds very little, what about simply removing the call and move it to estimatorComplementary() instead?

It is also questionable if zrange should be a member of sensorData_t at all. zrange seems to be the only sensor on an expansion deck that is included in the struct.

ataffanel, 2019-10-31:
I agree, zRange should most likely be pushed, the same way the LPS measurements are pushed in the estimator.

krichardsson, 2019-10-31:
There is also a position member in the sensorData_t struct that seems to be unused. Should also be removed.

closed 2019-11-01 by krichardsson