bitcraze/crazyflie-firmware #320: Crash in the Kalman filter

krichardsson, 2018-05-16:
Sometimes the kalman filter asserts/crashes with the following log entry
SYS: Assert failed at src/modules/src/estimator_kalman.c:845

AlexisTM, 2019-02-26:
What if it is caused by the ambiguity of having anchors in the same plane and as it is randomly fired, there is a chance that only one plane comes into play, thus breaking the asserts?

krichardsson, 2019-02-27:
@AlexisTM, unfortunately the problem also exists in a full system with 3D positioning.

krichardsson, 2019-08-21:
I have tried to recreate this issue in our flight arena with limited success. I only managed to get an assert in the kalman filter to fail one time (https://github.com/bitcraze/crazyflie-firmware/blob/3112896bdf49edba9caf5c702a5be9a5bfb38267/src/modules/src/kalman_core.c#L268).
We need some way to trigger this problem more often to be able to fix it.

krichardsson, 2019-08-21:
Finally I found a way to reproduce the problem fairly consistently. I  used our system test rig with 10 Roadrunners to increase the probability of hitting the problem, but it should work with a CF as well.

1. Turn on the LPS system in TDoA3 mode
1. Turn on the device(s)
1. Turn off the LPS system
1. Check the estimated position and wait for it to be capped at 100 m
1. Turn on the LPS system again
1. Sometimes the device(s) crash at this point

I suspect that the problem is related to the capping of position and velocity on lines https://github.com/bitcraze/crazyflie-firmware/blob/3112896bdf49edba9caf5c702a5be9a5bfb38267/src/modules/src/kalman_core.c#L937-L945

krichardsson, 2019-08-22:
Hopefully fixed by #458

closed 2019-08-22 by krichardsson