bitcraze/crazyflie-firmware #330: Adding compile flags for easier Crazyflie configure

ataffanel, 2018-06-13:
We have a couple of settings and work that could use compile flag to be enabled more easily.

Lets use this ticket to group commits that adds configuration compile flag for recent work.

ntamas, 2019-08-14:
How about using a `Kconfig`-style configuration framework (similar to what is used in the Linux kernel, `buildroot` and many other projects) to generate a `config.h` before compilation? There are many other parameters in the source code that are worth tuning (currently we are modifying the assumed weight of the drone as well as the velocity limits in the PID controller manually before compilation), and having a `make menuconfig`-style setup could definitely make things easier. You could also add switches for "experimental" features that you don't want to include in the stock firmware; users who are tech-savvy enough could then turn these features on and provide feedback.

ataffanel, 2019-08-15:
I have been thinking about kconfig-style configuration system for a long time but I am not even sure where to start looking in order to implement it (ie. do we make our own or do we re-use what the linux kernel already has).

There is also cmake that would be a very good things to integrate to make the build system cleaner more configurable and more cross-platform.

Both can be combined, this is what the Zephyr project is doing.

ntamas, 2019-08-15:
I have used [a Python implementation of Kconfig](https://github.com/ulfalizer/Kconfiglib) in another project of mine and it works (if you don't mind that the default color scheme is fugly). Incidentally, that other project is also using CMake as a build system so it's definitely a workable alternative.

I might take some time to look into this in the (not so near) future with the CF firmware.

closed 2021-08-30 by jonasdn