bitcraze/crazyflie-firmware #333: Multiple Loco Deck in TDOA3

AlexisTM, 2018-06-15:
The TDOA 3 system is awesome! Yet, to solve the yaw calibration at boot time, I plan to use 2 or 3 loco deck on a bigger frame. That way I can compute the yaw angle.

For this, I need a unique ID on each loco deck. Is a unique ID available?

krichardsson, 2018-06-20:
In general the crazyflie does not support multiple decks of the same type, so the short answer is no.

From a hardware point of view it is possible (with some slight modifications), the DW 1000 drivers supports multiple modules if needed while the TDoA/TWR/position estimation code does not, so you would have to implement this in one way or another to make this idea a reality.
It is possible that you could get some useful yaw information, but there would also be a lot of noise to handle.

Another option is to implement an Angle of Arrival deck with two DW1000 modules that are fed from the same clock. With this setup you can make sure the timers in the DW1000 modules are running in sync and you would get very precise timing information and better data. Of course this would also require HW design and manufacturing :-)

closed 2018-06-20 by krichardsson

AlexisTM, 2018-06-20:
I was thinking of using the loco deck outside of the crazyflie environment, on rovers (70x70cm) and writing myself the drivers to use the multiple decks, compute the angle and use it. 

To achieve it, I do need some way to differentiate the serial streams between the decks. For that, I wonder if there is a hardware (not hardcoded) unique ID, a chip ID, STM32 serial number,...