bitcraze/crazyflie-firmware #335: Controller mellinger yaw direction reversed

kyuhyong, 2018-06-19:
I tried to compare using PID and Mellinger for controller and found their yaw direction is opposite. 
In cfclient, right yaw commands gives positive deg/s while left yaw gives negative deg/s
with PID controller, cf2 turns right when I commanded right yaw however
with Mellinger controller, cf2 turns left for the same command.
Is this intentional? 
I tried to modify some code in the mellinger.c but changing the input does not help solving this issue. 
Thanks,

Kyu

krichardsson, 2018-06-20:
I think I'll pass this one on to @whoenig
Thanks!

whoenig, 2018-06-20:
This is not intentional and sounds like a bug to me. The mellinger controller was more heavily tested for operation in a mocap, so this might be related to the slightly separate manual flight mode. I'll take a look.

kyuhyong, 2018-06-27:
I applied the fix and tested with my drone today. 
It seems the issue is gone with both CONTROLLER: Using controller 2 or 1
Thanks,
Kyu

closed 2018-06-27 by kyuhyong