bitcraze/crazyflie-firmware #340: Fail to compile Arch Linux

snyderth, 2018-06-27:
I am trying to build the firmware on arch and every time I `make compile` or `make build` it returns the following error:
`In file included from **src/modules/src/crtp_commander_generic.c:35:
/usr/arm-none-eabi/include/math.h:133:12: error:** expected identifier or '(' before '**double**'
extern int isnan (double);`
'**src/utils/interface/num.h:41:28: error:** expected ')' before '?' token
#define isnan(n) ((n !- n) ? 1: 0)`

I just updated my local copy today. I'm building with config.mk options:
`PLATFORM=CF2
ESTIMATOR=kalman
LPS_TDOA_ENABLE=1
CLOAD=1`

snyderth, 2018-06-27:
Compiling with toolbelt works

closed 2018-06-27 by snyderth

ataffanel, 2018-07-02:
Good that it worked with the toolbelt/docker.

The original problem was most likely caused by a wrong version of GCC, the Crazyflie firmware should be compiled with [gcc-arm-embedded](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm). If you add it in your path you will be able to compile crazyflie-firmware directly from arch.