bitcraze/crazyflie-firmware #344: Is TDOA3 module test working now?

Huibean, 2018-07-03:
I am trying to exec runner.py there are a lot of ***.o file not found errors, I fixed that one by one by finally this so file stop me, I guess it's mock.c,
`make: *** [../../generated-test/module-test/tdoa3/tester.so] Error 1`

ataffanel, 2018-07-03:
Can you please describe a bit more what you are trying to do, how and what is failing in a bit more details?

Huibean, 2018-07-03:
Hi @ataffanel , I am trying to run tdoa3 module test
`➜  crazyflie-firmware git:(master) ✗ cd module-test
➜  module-test git:(master) ✗ cd tdoa3
➜  tdoa3 git:(master) ✗ ls
Makefile  input     mocks     run.sh    runner.py
➜  tdoa3 git:(master) ✗ make
cc -shared -o ../../generated-test/module-test/tdoa3/tester.so ../../generated-test/module-test/tdoa3/lpsTdoaTagEngine.o ../../generated-test/module-test/tdoa3/outlierFilter.o ../../generated-test/module-test/tdoa3/mocks.o
ld: warning: ignoring file ../../generated-test/module-test/tdoa3/lpsTdoaTagEngine.o, file was built for unsupported file format ( 0x7F 0x45 0x4C 0x46 0x01 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 ) which is not the architecture being linked (x86_64): ../../generated-test/module-test/tdoa3/lpsTdoaTagEngine.o
Undefined symbols for architecture x86_64:
  "_getAnchorCtxForPacketProcessing", referenced from:
      _callTdoaEngineProcessPacket in mocks.o
  "_tdoaEngineProcessPacket", referenced from:
      _callTdoaEngineProcessPacket in mocks.o
  "_tdoaEngineSetAnchorPosition", referenced from:
      _callTdoaEngineProcessPacket in mocks.o
  "_tdoaEngineSetRemoteRxTime", referenced from:
      _callTdoaEngineProcessPacket in mocks.o
  "_tdoaEngineSetRxTxData", referenced from:
      _callTdoaEngineProcessPacket in mocks.o
  "_tdoaEngineSetTimeOfFlight", referenced from:
      _callTdoaEngineProcessPacket in mocks.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [../../generated-test/module-test/tdoa3/tester.so] Error 1
➜  tdoa3 git:(master) ✗`

ataffanel, 2018-07-05:
I am not sure how this is supposed to work since it was written by Kristoffer as an experiments and he is currently in vacation.
I get a different error when trying to build though. You are using clang so I assume you are on Mac, the code has always been compiled with GCC and maybe this is causing part of your problems.

krichardsson, 2018-08-03:
The module test is highly experimental and should probably not have been committed yet. The idea is to compile and link a few C files from the firmware source code into a lib that can be called from python on a PC. The files that are tested should make up a "module", for instance the TDoA 3 engine, that can be tested on a higher level than unit tests do. 
I will remove it or clean it up and make it an integral part of the automated tests.

Huibean, 2018-08-06:
@ataffanel @krichardsson ok, get it

closed 2018-08-06 by Huibean