bitcraze/crazyflie-firmware #349: How to fix the orientation of drone?

kyuhyong, 2018-07-17:
Hello everyone,
Whenever a drone is powered up in inclined floor, it tries to calibrate its orientation to the surface as if the drone is in flat ground which makes taking off in tilted angle. 
Is there a way to let the kalman estimator always fixes its orientation to its body frame rather than global fixed frame?

ataffanel, 2018-08-21:
The only way would be to use the accelerometer to st the initial estimate of roll/pitch in the Kalman filter. This is possible but not implemented at this point, if you are interested to implement it as a compile flag a PR is welcome :-).

One thing that will most likely need to be modified is the roll/pitch zero reversion: it is currently assumes that the Crazyflie stops flat and so roll/pitch slowly reverts to zero when the Crazyflie is not flying. This is happening there: https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/estimator_kalman.c#L776-L779. If you initialize the Crazyfle on a non flat surface the zero reversion should either be set to 0 or the reversion should be implemented to go towards the accelerometer-measured roll/pitch.

benkuper, 2018-09-19:
+1 for the possibility to reset kalman estimation and still be able to retrieve "absolute" orientation of the drone.
I'm implementing a safe behavior where the drone should not be taking off when upside down (already implemented in the firmware), but if the drone is recalibrated when upside down, then it will think its roll is 0.

AlexisTM, 2019-02-27:
I would instead vote for a **Precalibration** saved in the EEPROM with 3 options: 
* Use precalibration parameters;
* Use auto calibtation at boot;
* Use auto calibration at boot and if it fails in 10 seconds, use precalibration parameters

AlexisTM, 2019-03-06:
@kyuhyong I worked on making the crazyflie saving its calibration into the EEPROM if it succeeded and loading it from the EEPROM if calibration failed and the EEPROM calibration is valid.

https://github.com/umdlife/crazyflie-firmware/pull/2/files

I guess your problem is much deeper than that. The CF horizon is set by fusing the sensors into the Kalman filter. Therefore the horizon is zeroed at boot (Kalman initialization). It will likely be hard to take that calibration out of the current system and use a level horizon similar to the PX4 Firmware (level horizon).

jonasdn, 2021-09-06:
Is this related to #547?

jonasdn, 2021-09-07:
For the Crazyflie it is a hard mandate to start from a flat surface.

closed 2021-09-07 by jonasdn