bitcraze/crazyflie-firmware #350: loco_deck without the crazyflie

AlexisTM, 2018-07-18:
We have many applications where we require TDoA localization. 
As the loco_deck is only connections to the DWM1000, we need to hook it to a microcontroller. 
Do you plan on making a breakout board containing an STM32 for the loco deck?

krichardsson, 2018-08-03:
We are looking at something like this, most probably it would be something like a CPU + IMU + DWM1000 on one board. Unfortunately the price would be in the same region as for a node.

AlexisTM, 2018-08-03:
If it is a standalone device with an IMU, 150$/180$ could fit the budget.

Note: We are currently using the [Pozyx](https://www.pozyx.io/store/detail/4) but we are looking for alternate solutions for the following reasons:
* Not open-source
* TDOA (passive tags) is not on their roadmap
* The IMU is not configurable (as not open-source we cannot add it)

krichardsson, 2018-08-06:
Thanks for the feedback!

closed 2018-08-06 by krichardsson

AlexisTM, 2018-08-09:
@krichardsson There is a last feature that could be nice to add on a loco deck V2: AoA localisation by putting 2 or 3 DWM1000 antennas. 

We can then remove the fact that we need to put the crazyflie/other robot aligned with the localisation system as first approximation.

AlexisTM, 2019-01-24:
For anybody interested: this is in early adopter, under the name of Roadrunner.

krichardsson, 2019-02-05:
The Roadrunner is now available in our store. https://store.bitcraze.io/collections/positioning/products/roadrunner