bitcraze/crazyflie-firmware #365: Base Thrust / Scaling for more powerful motors

yashm, 2018-08-28:
Enhancement request:
This is per a phone conversation with Tobias on Aug 21, 2018.

Request to add a feature to calibrate thrust mapping after changing to more powerful motors to the Crazyflie 2.0.

[Current mapping exists here](https://github.com/bitcraze/crazyflie-firmware/blob/499f0184c9de64397b444e3eec8db51cf859b753/src/drivers/src/motors.c#L213), but would be great if "base-thrust" were available as a parameter like it is in the _cfclient_ GUI.

ataffanel, 2018-08-31:
Adding parameters is the easy part, though I am wondering about what other changes is needed there:
 - How would the new mapping looks like, similar to the current one or with a different formula?
 - What variables would need to make accessible as parameters?

yashm, 2018-09-06:
1. The mapping would be similar to what it is now, but by setting the max thrust to a value far greater than any expected value (~3kg) and then following point 2 below, will make the software more configurable. Doing this will reduce the resolution, but with a value of 3kg, it will still be 0.046g per bit.

2. At a minimum, the variables that can be made editable would be: 
    - Mass of the robot.
    - The preferred way would be to set thrustMin and thrustBase as percentages of the mass of the robot.
eg: thrustMin = 5% of the weight of the robot.
      thrustBase = 100% of the weight of the robot.

yashm, 2018-09-10:
Maybe even consider making the robot mass configurable through the config.mk file?

jonasdn, 2021-09-07:
This will be part of getting the Bolt and BigQuadDeck out of early access

closed 2021-09-07 by jonasdn