bitcraze/crazyflie-firmware #380: Remapping LED-ring deck pins for the top mount

bgromov, 2018-11-12:
**Platform**: Crazyflie 2.0
**Firmware**: 2018.10-rc1-3 (modified)
**Decks**: Flow (v1, Rev. D), LED-ring (Rev. B)

We are trying to build an adapter for the LED-ring deck that would allow to mount it on the top of Crazyflie instead of on the bottom. Crucially, we also need the Flow deck in the same setup, thus simply swapping the pins from left to right would not work.

So far what we came up with is remapping (and re-wiring) of the `IO_2` (`PWM_DOWN`) to `IO_1`, and `IO_3` (`PWM_FRONT`) to `TX2`. That means we had to use `TIM4_CH3` (with `DMA1`, channel 3, stream 7) instead of `TIM3_CH2` (`DMA1`, channel 2, stream 5). Here are the changes we made: https://github.com/bitcraze/crazyflie-firmware/compare/master...bgromov:led-top-mounted

Both, the LED deck and the Flow deck seem to work fine, but there is a very weird issue with the motor 4: it doesn't spin normally anymore and the drone is very unstable. The "Propeller test" in `cfclient` reports an error like this:

```
STAB: Acc noise floor variance X+Y:0.000808, (Z:0.001060)
STAB: Motor M1 variance X+Y:0.413310 (Z:0.810043)
STAB: Motor M2 variance X+Y:1.368914 (Z:3.055007)
STAB: Motor M3 variance X+Y:0.266747 (Z:0.435409)
STAB: Motor M4 variance X+Y:7.209903 (Z:11.663726)
STAB: Propeller test on M4 [FAIL]. low: 0.0, high: 2.50, measured: 7.20
```

If I understand correctly, this is the variance of the current (?) used by the motor. Sometimes the reported value is much higher than this, e.g. > 25; sometimes, very rarely, the propeller won't spin at all (though the value is low and the test succeeds). Interestingly enough, with the vanilla firmware and the LED-ring deck mounted where it should be the Propeller test fails on M4 anyway, but usually with a lower value (around 4-5) and the drone flies fine.

We also tried this test on another Crazyflie and it consistently fails there on the M4 as well. When the LED-ring deck is mounted on the bottom, the test fails on M4 but the drone flies fine; when mounted on the top with our adapter it fails with slightly higher values, but then couldn't take off properly.

Searching through the sources shows that the `TIM4_CH3` is not really used anywhere (it is only referenced by the structure [`DECK_IO1_TIM4`](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/drivers/src/motors_def_cf2.c#L381)). So I am quite puzzled what else it could interfere with.

Any ideas what could be the problem?

Any help is highly appreciated.

Thanks,

Boris

tobbeanton, 2018-11-13:
Thanks for a well described problem! I'm doing the assumption that the flow deck is mounted down.

I think the problem is that even though TIM4_CH3 is not used they can not share the same TIM4 peripheral as they need different types of configuration.

If you are not using the white forward looking LED it should be enough cabling 3 wires (IO_2, VCOM, GND) and forcing the deck. Then you can't use the pin-header as the holding structure and you would have to attach the battery and LED-ring with other means.

PS. The propeller test is measuring the vibrations using the accelerometer. DS

bgromov, 2018-11-13:
Thanks for the quick reply!

So, what pin can I use instead of `IO_2`? I was using `TX2` for the Headlights, although it works, I now see that it is used by the Flow deck (`RESET`). Another problem is that the DMA Stream 0 on `TX2` with `TIM5_CH3` is already taken by i2c. The `RX2` seems to be used by the Flow deck as well.

Are there any other options?

Thanks

tobbeanton, 2018-11-14:
The headlights could probably be mapped to some other pin such as RX2 or TX2 (that the buzzer deck uses). In [this](https://www.hackster.io/krichardsson/light-paint-with-a-drone-d050af) hackster project which uses the Flow and the LED-ring the headlight is just not connected (cut).
IO_2 is used for the WS2112 LEDs and I think will be hard to move but it is also not connected on the Flow deck so this is fine. The problem is IO_3 which is PWM on LED-Ring and ChipSelect on Flow. I would try moving it to RX2 or TX2.

bgromov, 2018-11-14:
That worked! I didn't look into the Flow deck schematics carefully enough and thought that the `IO_2` is actually used.

Thanks for your help!

closed 2018-11-15 by tobbeanton