bitcraze/crazyflie-firmware #389: Pre-programmed Waypoint Navigation?

NicksonYap, 2019-02-06:
Hi,

I assume pre-programmed waypoint navigation is not yet supported, so I wonder:

Is Crazyflie 2.0 on-board memory capable/suitable of storing on-board waypoints
Or are external hardware like SD Cards recommended?

I'm planning to have RGB, XYZ Yaw commands included and likely to be at 10Hz (or more)

What are the related CF firmware functions/api that we're supposed to extend/use to implement waypoints?

Storage functions, Custom Crazyflie commands etc

How/What is the proper way to implement Custom CrazyRadio commands?

I'm familiar with ArduPilot, and have added custom MavLink commands before.
CF should be easier

Side question:

I'm looking into pre-programmed waypoints in particular because I might be having issues controlling more than 3 drones simultaneously. I was transmitting RGB & XYZ commands at 10Hz.
What is the approximate rate of commands CrazyRadio is capable of?
The RSSI seems strong and at close distance using directional antenna

krichardsson, 2019-02-07:
Hi!

There is a High level commander added by the Crazyswarm contributors that might be what you are looking for. It has been contributed in a number of PRs but check out the #293 issue for an overview.

The high level commander takes polynomials as input, describing the trajectory to fly. It gives full control but it is unfortunately a bit tricky to generate the polynomials. There is a simple "goto" function though that can be used to simply go to a specific coordinate. 

The high level commander can be accessed through the python API, it is possible to up load a trajectory that is started at a later point in time. There is an example in 
https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/autonomous_sequence_high_level.py

NicksonYap, 2019-04-26:
@krichardsson sorry for replying so late

I was aware of the navigation commands on Python

The real issue here is how to save a series of navigation commands on-board the crazyflie
And then playback on the crazyflie without depending on Crazyradio for navigation commands, but only start/stop/pause commands

krichardsson, 2019-04-29:
The High Level commander is implemented in the firmware, the python lib is just a wrapper to use it from a computer. All the action is in the firmware so it should be useful for what you want to do. 

You can find an example of how we have used it at IROS last year https://github.com/bitcraze/crazyflie-firmware-experimental/tree/iros-2018-autonomous where we had a preprogrammed trajectory.

NicksonYap, 2019-04-29:
As a kickstart, did you guys pre-program it in external SD card or internal flash, etc?
Able to quickly give any link to line of code that is relevant?
Then i can start digging deeper

krichardsson, 2019-04-29:
Everything happens in https://github.com/bitcraze/crazyflie-firmware-experimental/blob/iros-2018-autonomous/src/modules/src/app.c
Other changes are mainly to expose functions that are static. 

We have also started to work on the demo for ICRA in a couple of weeks. It is an extension of the IROS demo but fairly similar. https://github.com/bitcraze/crazyflie-firmware-experimental/blob/icra-2019/src/modules/src/app.c. This might actually be a better branch to look at since it is more up to date with master in crazyflie-firmware.

NicksonYap, 2019-04-29:
Thanks!
Exactly what I needed! 👍 

The ICRA19 version seems to mention lighthouse deck & IROS18 doesn't.
Just to confirm, the IROS one uses UWB deck right? (I'm intending to use UWB deck for now, and swap to lighthose later)

Any suggestions on how the Sequence array should be stored externally? 
I mean it doesn't make sense for us to flash say tens or even hundreds of Crazyflies everytime we change the sequence.
I don't think storing in eeprom will do any good:
https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/write-eeprom.py

Is there an example for reading from SD card in Crazyflie?
I don't mind storing strings in a txt file, parsing in CF and load it just like in `app.c`

bitcraze-ci, 2019-04-30:
Hi!

Yes, we will use the Light house for the ICRA19 demo, but most of it should
work with the LPS as well. You might want to not use the mellinger
controller, unless you have a very good position estimate, it is more
sensitive to outliers. You also want to disable the Lighthouse deck by
uncommenting the lines with defines for DISABLE_LIGHTHOUSE_DRIVER in
lighthouse.c to avoid building the lighthouse deck binary into your flash
image (it would hurt but flashing takes longer).
We used LPS at IROS18 so this should be all good.

>>> Any suggestions on how the Sequence array should be stored externally?
Not really.
In CrazySwarm (at USC) they upload the trajectories via radio after boot.
In theory the SD-card could be an option but there is a known issue with
the SD-card and LPS deck (
https://github.com/bitcraze/crazyflie-firmware/issues/270).

br
Kristoffer

-- 
----------------------------
Kristoffer Richardsson
Bitcraze AB

E-mail: kristoffer@bitcraze.io
Phone: +46 733 467 764
Skype: kristoffer.richardsson
Web: www.bitcraze.io
Github: krichardsson
Keybase: krichardsson

J-Durotifa, 2019-06-24:
Hi guys,
Have you made any progress with uploading waypoints via radio? 
I don't want to go the route of flashing crazyflies for every waypoint change and I'd prefer not to use the polynomial trajectory because of its limitations.
I'm currently looking into this and found this thread, but I don't know what files to look for in order to achieve this, so I was wondering if you could help.

NicksonYap, 2019-06-24:
@J-Durotifa So far flashing crazyflies is the most direct way

I'm starting to consider looking into using a separate microcontroller (Arduino) as a deck to handle storage of waypoints/commands 

As if the Microcontroller/Arduino Deck is the python api controlling the Crazyflie (except it is via serial, or I2C)

@krichardsson are there any hardware interfaces (serial/I2C) for sending commands to Crazyflie?
And also possibly relay Crazyradio messages to external microcontroller?

I feels like we'll need do implement ourselves in `app.c`

ataffanel, 2019-06-25:
@J-Durotifa we are not looking at uploading waypoint via radio, though this is something that could be added to the Crazyflie firmware high-level-commander code the same way polynomials are currently implemented and played back. This modification would not require major change in architecture as long as you do not want to store waypoints in the Crazyflie. 

@NicksonYap The way the communication is currently handles only allows for one CRTP 'master'. It would be possuible to control the Crazyflie from a serial port (this has been done in the forum: https://forum.bitcraze.io/viewtopic.php?f=2&t=3435&p=16772&hilit=lte#p16720), though as soon as you start communicating over UART the radio link is disabled.

If you only need some memory, and you need to make change in the Crazyflie firmware anyway, the approach of controlling the Crazyflie from an external MCU might not be the best way anyway. The Arduino could be used as a serial or i2c storage though. However the best might be to add an SPI memory or use the SD-card as you suggested earlier. Unfortunately I just had a look and it seems that the SD-card deck driver is very much specialized into implementing log, we need to break it down and add an API to access it from other part of the firmware: we have fat filesystem handling so it would be natural to allow other code in the Crazyflies to read and write files.

As for the architecture. If you make a deck you most likely want to make a [deck driver](https://wiki.bitcraze.io/doc:crazyflie:api:firmware:deck) and then you need an ```app.c``` that implements the autonomous functionality you need (very similar to what we did for ICRA, the drones where able to autonomously fly sequences and come back to land and charge).

As a side note: there is some free flash in the Crazyflie (at least 500KB), this could be used for trajectory storage. The bootloader can be used to program custom region of flash or some programming functionalities could be added in the firmware.

krichardsson, 2019-08-12:
@NicksonYap @J-Durotifa I'm reviewing issues, could we close this one?

NicksonYap, 2019-08-12:
Yup, I think it's very well answered

closed 2019-08-13 by krichardsson