bitcraze/crazyflie-firmware #391: Motor control battery compensation should be capped

ataffanel, 2019-02-06:
With the nRF firmware bug bitcraze/crazyflie2-nrf-firmware#28 the crazyflie sets the thrust of the motor to full thrust as soon as the motors are enabled. This is caused by the battery voltage comprensation formula when the battery voltage is set to 0.

It is never OK to set the thrust to 100% so the battery compensation factor should most likely be capped within reasonable value to mitigate the effect of such bugs.

AlexisTM, 2019-03-08:
The battery is supposed to be 4.2V charged, 3.7V mid-charge and 3V discharged (LiPo).

Therefore, I would set a threshold of disabling the voltage compensation under 2V. A 2 volts (1S) LiPo is permanently damaged already.

jonasdn, 2021-09-06:
What would be a suitable cap percentage? Do we know? How do find out if not?

And why is it never ok to set the thrust to 100%?

commit cb61f8d9 (2021-09-06): motors: Sanity check supply voltage in compensation

Closes #391
PR #847 (2021-09-06): motors: Sanity check supply voltage in compensation
Closes #391
commit d7d9c3df (2021-09-06): motors: Sanity check supply voltage in compensation

Closes #391
closed 2021-09-06 by jonasdn