bitcraze/crazyflie-firmware #392: TDOA3 is drifting away if it fails to locate.

AlexisTM, 2019-02-18:
**Bug:**

The roadrunner position will drift away quickly when out of range of the anchors. The drift is much faster than the covariance.

**Reproduce:**

- Configure the roadrunner anchors
- Experience that you have a good positioning inside the covered area
- Move away from the covered area (or shut down down anchors?)
- The position will drift away very quickly (much faster than the covariance)

AlexisTM, 2019-02-20:
The covariance is understimated, the zero process covariance certainly plays a role.

https://github.com/bitcraze/crazyflie-firmware/blob/43b6c25db2ee8fff84d97f4ffef1b5c1f2a63e3d/src/modules/src/estimator_kalman.c#L225-L227

krichardsson, 2019-02-27:
Unfortunately this is not my area of expertise.
@mikehamer, do you have any input on this?

mikehamer, 2019-02-27:
These process noises are added to the integrated states of position, velocity and attitude. Although these direct terms are zero, noise still increases the uncertainty on position, velocity and attitude through the noise on acceleration and body rate (procNoiseAcc and measNoiseGyro respectively). These direct terms are set to zero, since in a perfect world it doesn't make physical sense for the integrated states to be directly affected by a noise term.

I agree though that perhaps a small value may improve estimation,  particularly using less-than-perfect localization like UWB, which has very non-gaussian disturbances, or may improve the estimator's ability to handle situations where the state estimate is far from the UWB measurement (I think you asked about such a situation in a different issue?). Honestly, I've not tried it. If you find a setting that improves flight, please don't hesitate to submit a pull request :-)

krichardsson, 2019-02-27:
Thanks @mikehamer!

mikehamer, 2019-02-27:
Ahh I also now read the original issue (and not just your first reply to it about the covariance).

Time-difference-of-arrival localization, eg. as used in the TDoA3 system, is extremely sensitive to noise if you are operating outside the area spanned by the anchor system, basically to the point of the system being almost unusable in such a case. This is inherent in TDoA localization and the maths behind it. For the best accuracy, anchors should be placed far apart, and tags should always be within the area covered by the anchors. 

Take a look at the paper [here](http://mikehamer.info/assets/papers/Ledergerber,%20Hamer,%20DAndrea%20-%20IROS%202015.pdf) for a plot of this sensitivity (Figure 3, left side).

This is separate to your note that position drifts away faster than covariance, but is something that will cause issues even in the best case.

AlexisTM, 2019-02-27:
@mikehamer Thank you very much for the information.

Note that I wrongly thought that the TDoA was drifting. The position is, in fact, drifting away even if there wasn't any TDoA measurement ever. I am used to PX4 and I took for granted that the position is not computed (set to 0,0) if there is no valid localization system (velocity or position measurement).

I will therefore close this issue because it is not linked to TDoA

closed 2019-02-27 by AlexisTM