bitcraze/crazyflie-firmware #396: Fix sign inconsistencies in pose

tobbeanton, 2019-02-19:
As described by Wolfgang:
For the negative signs and the power distribution: Unfortunately, there are many inconsistencies in the firmware. You can find a non-exhaustive list here: https://lib.dr.iastate.edu/etd/15575/ (Appendix A). Essentially, those signs compensate for those issues.

It would be good to try to correct this at some point.

knmcguire, 2020-06-26:
Here is the list from that paper:

- [ ] Body axis not aligned with aerospace coordinate system

- [ ]  vector accelerator meter-inputs are negative of gravitational vector (in complementary filter)

- [ ] Pitch angle is negated

- [ ] yaw rate is negated before entering mixer

- [ ] q body rate measurement is negated before q-rate PID computation

- [ ] Measured yaw angle is negated before transmission

- [ ] x-position pid controls the roll angle and y position controls the pitch angle

It is quite a lot, but probably it is just a matter of going through it all and make sure we get all the signs. Also documentation on how the coordinate system is defined in the crazyflie would be useful to add as well.

knmcguire, 2022-03-07:
This issue also refers to this: #884

knmcguire, 2022-09-06:
This is related to the bigger SI unit change discussions