bitcraze/crazyflie-firmware #398: TDOA measurement process

AlexisTM, 2019-02-25:
I would like to know more about how the position is computed/fused with the TDOA3 method. 

It seems to be: 
- Receive data from anchors (locodeck.c)
- Send tdoaMeasurement to a Queue (locodeck.c)
- Receive tdoaMeasurement from a Queue (estimator_kalman.c)
- For each tdoaMeasurement in the Queue compute the distance vs predicted distance and [it seems] to be only updating the covariance (estimator_kalman.c)

What are the next steps?

Indeed, the following code : [estimator_kalman#L988-1049](https://github.com/umdlife/crazyflie-firmware/blob/ffe84e1ef35435dd93d7a5877f2884ba34c60670/src/modules/src/estimator_kalman.c#L988-L1049) only computes the distance between two anchors and the error of that measure (difference to the predicted distance). Where are those measures fused together into an XYZ position?

I do think that the method to compute the error is lacking a few points:
- If the current position drifted away, the covariance for the predicted value is very high (if we did set `procNoisePos` to non zero value). Thus; the measurement error is very high because the predicted value is wrong. The measurement will be discarded. 
- The error does not take in account the `DLPS_2D_POSITION_HEIGHT` tag, which "should force the STATE[Z] to the `DLPS_2D_POSITION_HEIGHT` value.

krichardsson, 2019-02-25:
The main function that drives the kalman filter is [estimatorKalman()](https://github.com/bitcraze/crazyflie-firmware/blob/2019.02/src/modules/src/estimator_kalman.c#L364-L541), where the queues are emptied (updating covariance), the state is updated and the position calculated. It is called from [estimator.c](https://github.com/bitcraze/crazyflie-firmware/blob/2019.02/src/modules/src/estimator.c#L62) which is called from the main [stabilizer loop](https://github.com/bitcraze/crazyflie-firmware/blob/2019.02/src/modules/src/stabilizer.c#L179).

@mikehamer implemented the kalman filter, maybe he can share some light on the details?

I have added the outlier filter in outlierFilterVaildateTdoaSteps(). The intention is to discard samples that seems to be too far away from the estimated position, assuming they are corrupt due to interference or similar. It has a "back off" functionality that accepts all samples to let the kalman filter converge to the correct position again if it seems as the estimated position is wrong.

The 2D positioning is sort of a hack that assumes the Tag is at constant Z. If the Tag is at the correct height I assume the error calculation should be OK?

mikehamer, 2019-02-25:
Hi Alexis,

The measurement error is used to update the state estimate in the "stateEstimatorScalarUpdate" function via the computed H matrix (https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/estimator_kalman.c#L1043). This step completes the Kalman filter's measurement update.

Does this answer your question about the Kalman filter's operation? I can't comment on the outlier checking since I wasn't involved there, but your points make sense. I often use a timeout for discarding measurements (e.g. if no measurements have been accepted in the last 0.2 seconds, then accept the current measurement).

AlexisTM, 2019-02-27:
I also noticed an overshoot on the measurements on the position output of the Kalman filter.
In the test setup, the roadrunner is positionned on a box at 31cm height and built with a the fixed height flag of 0.31. 

The box is moved quickly of around 4 meters forward in the x direction. The X value overshooted up to 5m.

I therefore would like to decouple the TDoA measurements of the IMU and filter it on a higher level (ROS). Do you have one task decoupled for testing purpose or should I do mine?

[image]

mikehamer, 2019-02-27:
Judging by the position returning to the correct value (I think?), my guess is that this isn't to do with the positioning system, and is rather due to the assumed dynamics of the vehicle.

The Kalman filter integrates accelerations to give velocities, which are integrated to give position. If you move the box and stop it abruptly, this behavior is likely not captured very well by the model and hence the velocity will continue to be integrated and affect position. This would particularly be the case if the accelerometer readings (showing the abrupt stop) saturate below their true value.

I don't know anything about the roadrunner @krichardsson, are you using the same Kalman filter as for the Crazyflie?

Try moving and decelerating the box slowly, do you see similar behaviour?

krichardsson, 2019-02-27:
@mikehamer, yes the Roadrunner currently has the same Kalman filter as the Crazyflie. I guess it would make sense to modify it based on the dynamics of the vehicle it is attached to?

By the way, It has the same sensors as Crazyflie 2.1. We should probably measure the noise of the sensors (should be better than CF2.0) and adjust the numbers in the Kalman filter.

AlexisTM, 2019-02-28:
This is indeed likely to be due to the dynamics of the system.
What we are looking forward to is to decouple the IMU arond the TDoA system. That way, we can filter the data from ROS (robot_localization) and have a behaviour which is not coupled to drone dynamics. 

@mikehamer Moving slower reduces the issue; With the AGVs driving, the overshoot of the localization system is making the AGV slowing down faster than expected (because the localization system reaches the goal faster than the real AGV). 

The other issue we have (see https://github.com/bitcraze/crazyflie-firmware/issues/349#issuecomment-467779695) is that the Lidar of the AGV is inducing too high vibrations in the AGV frame for the roadrunner to calibrate, freezing the kalman filter. For the tests, we had to start the roadrunner on a flat surface than place it on the AGV.

AlexisTM, 2019-03-06:
The issue of the vibrations is worked around by implementing a calibration saved into the EEPROM by extending the configblock. 

https://github.com/umdlife/crazyflie-firmware/pull/2

We are still looking into a way to decouple the TDoA measurement process from the Kalman filter.

krichardsson, 2019-03-06:
@AlexisTM , decoupling the TDoA measurement process from the Kalman filter sounds interesting. Could you please open a new issue for this where we can discuss use cases, design and implementation.

krichardsson, 2019-08-12:
@AlexisTM, I'm going through issues. Could we close this one?

AlexisTM, 2019-08-12:
Yes! This can be closed off. 

Sorry for letting it pending for so long. 

For anybody interested, we are publishing the TDOA measurements through the serial port over MAVLink using the following firmware:

https://github.com/umdlife/crazyflie-firmware/tree/u2_feature_custom_kalman

The MAVLink XML definition is the following:

https://github.com/umdlife/roadrunner_mavlink

Eventually, you can as well use the multilateration_tdoa library which locate the tag defined as a minimisation optimisation problem.

https://github.com/alexistm/multilaterationtdoa

closed 2019-08-12 by AlexisTM