bitcraze/crazyflie-firmware #399: Roadrunner: Sometimes the estimator does not update the IMU/Position

AlexisTM, 2019-02-25:
From time to time, both the IMU and the TDOA position stays at the initial values. 

- IMU is at {x:0, y:0, z:0, w:0}
- TDOA3 position is at {x:0.5, y:0.5, z:0}

It does not not update even after waiting a full minute.  

This bug is random. 
I wonder if it could be due to some external factor such as starting the roadrunner and receiving a TDoA packet at a wrong timing which would crash the EKF or if it is due to the IMU bug that would make the kalman not starting at all while the IMU is not ready (and never is)

AlexisTM, 2019-02-25:
The precedent idea (TDoA packet at wrong timing) seems unlikely due to the many guards `ASSERT(isInit);`. 

I kind of reproduced it by having a weak TDoA reception (being in another room with a door closed than the one with TDoA installed).

I have the feeling that the EKF failure happens when we receive packets but not enough to compute the position. NOTE: When I finally have a position (or have a position drifting), after some time, it happens that I get a NaN (2 to 3 last messages have a NaN value) followed by "USB TIMEOUT".

NOTE: I am using [umdlife/crazyflie_ros](https://github.com/umdlife/crazyflie_ros) and the firmware with 

```makefile
# Watchdog disabled and:
CFLAGS += -DLPS_2D_POSITION_HEIGHT=0.31
CFLAGS += -DLPS_TDOA3_ENABLE=1
CFLAGS += -DLOCODECK_NO_LOW_INTERFERENCE
```

Last positions, logging at 100Hz

```yaml
---
header: 
  seq: 19844
  stamp: 
    secs: 1551103906
    nsecs: 366745319
  frame_id: "map"
pose: 
  position: 
    x: -98.9362716675
    y: -99.8890838623
    z: 0.27147346735
  orientation: 
    x: 0.898630559444
    y: -0.361439675093
    z: 0.0513428859413
    w: 0.243286848068
---
header: 
  seq: 19845
  stamp: 
    secs: 1551103906
    nsecs: 378047211
  frame_id: "map"
pose: 
  position: 
    x: nan
    y: nan
    z: nan
  orientation: 
    x: 0.898627281189
    y: -0.361448466778
    z: 0.0513433814049
    w: 0.243285596371
---
header: 
  seq: 19846
  stamp: 
    secs: 1551103906
    nsecs: 387359084
  frame_id: "map"
pose: 
  position: 
    x: nan
    y: nan
    z: nan
  orientation: 
    x: 0.898632228374
    y: -0.361445993185
    z: 0.0513430833817
    w: 0.243271037936
---
header: 
  seq: 19847
  stamp: 
    secs: 1551103906
    nsecs: 396690956
  frame_id: "map"
pose: 
  position: 
    x: nan
    y: nan
    z: nan
  orientation: 
    x: 0.898632526398
    y: -0.361456185579
    z: 0.0513554625213
    w: 0.243252307177
---
```

AlexisTM, 2019-02-25:
NOTE: 

The EKF seems to be failing reliably when starting the crazyflie with movement or without being flat. 

There is no error message in the console.

krichardsson, 2019-02-25:
It might be that your sensors are not calibrated?
you can read more about the [LED blinking patterns here](https://www.bitcraze.io/getting-started-with-the-crazyflie-2-0/#leds) to see if the sensors has been calibrated. The Roadrunner/CF must be flat and still to calibrate the sensors.

krichardsson, 2019-02-25:
There is also a known issue in the kalman filter (#320) that causes it to crash. It usually causes the Crazyflie to freeze or restart.

AlexisTM, 2019-02-27:
@krichardsson Indeed; The calibration is the problem. The Crazyflie is not calibrating on the robot because the Lidar is inducing vibrations in the frame. Therefore, until the IMU is ready, the Kalman filter is "on hold".
I guess I will have to look into a solution of precalibration of the IMU and saving it into the flash.

AlexisTM, 2019-03-06:
@krichardsson I did a quick fix which

AlexisTM, 2019-03-06:
@krichardsson I did a quick modification to fix this. The PR is for my fork and includes some unwanted commits for the Bitcraze repo. Yet, the diff is valid to look into it. 

https://github.com/umdlife/crazyflie-firmware/pull/2

I would love that someone can double check if there is something I overlooked such as a `configblock` size hardcoded for the EEPROM location that would risk an overflow or so. If it's valid, I could create a PR for the roadrunner if the PLATFORM is tag.

Additional changes I would do for a PR:
- The accelerometer calibration was not required.
- I guess I moved too much to `configblock.h`
- It would be nice to have the choice to enable/disable/force the usage of the EEPROM calibration.

closed 2019-04-24 by AlexisTM