bitcraze/crazyflie-firmware #405: Uart1 and Uart2 are broken due to initialization without baudrate.

AlexisTM, 2019-03-13:
Adding `CFLAGS += -DENABLE_UART2` (or UART1) will break the software as the uartInit functions are called without the baudrate specified which is required for the function call.

https://github.com/bitcraze/crazyflie-firmware/blob/d8d92d1ac7dafe6921912666efbea0c696890b0b/src/drivers/src/uart2.c#L47

https://github.com/bitcraze/crazyflie-firmware/blob/741a7860aa1934fc74af679ba793187d4016c116/src/modules/src/system.c#L150-L155

krichardsson, 2019-03-14:
Seems as the uart init functions were modified at some point to support baudrate, but the calls in the ifdefs were not.
When fixed, ENABLE_UART1 and ENABLE_UART2 should be added to the .travis.yml file to avoid similar problems in the future.

AlexisTM, 2019-03-15:
I guess the default of UART1 is 9600 (available in `uart1.h`) and I would propose uart2 to be 115200 by default (It is the roadrunner uart)

krichardsson, 2019-03-15:
Fixed!

closed 2019-03-15 by krichardsson