bitcraze/crazyflie-firmware #406: Option to choose orientation of drone at calibration

benkuper, 2019-03-14:
It would be nice to have an option to be able to choose how the drone is when calibrating, currently is default to x positive, it could be something like
X+, X-, Z+, Z-
It would allow to have in a performance context the drone to be facing the public (Z-) or the performer behind (Z+)

It think it should at least be defaulting to Z- , which is probably the most used (drone facing front, meaning we are looking at the LPS cube with X+ being on the right and Z positive being behind

krichardsson, 2019-06-18:
The initial yaw can be set through the kalman.initialYaw parameter. The initial yaw is used to initialize the estimator when the Crzyflie is not oriented in the positive X direction.

Initial Yaw is set in radians where:
0: facing positive X (default)
PI / 2: facing positive Y
PI: facing negative X
3 * PI / 2: facing negative Y

Note: initial yaw takes effect after the kalman filter is reset.

closed 2019-06-18 by krichardsson

benkuper, 2019-06-18:
Nice !