bitcraze/crazyflie-firmware #407: high level commander zero divide check

knmcguire, 2019-03-15:
I noticed that the high level commander in the python script had some problems (see bitcraze/crazyflie-lib-python#112), where it fell out of the sky if the same setpoint was used for the go_to function in a row.

I noticed that in crtp_commander_high_level, non of the functions who uses these setpoints (go_to()->plan_go_to()->piecewise_plan_7th_order_no_jerk()->poly7_nojerk() ), check if the duration is zero. This is important since in poly7_nojerk, the duration is the variable that divides another variable. So if the setpoints are the same, the duration to get to that setpoint will be zero and that complicates things.

There should probably be a check in the crazyflie-firmware that prevents this divide by zero from happening.

whoenig, 2019-03-28:
I think this can be closed now.

ataffanel, 2019-03-29:
Agreed, thanks for the PR!

closed 2019-03-29 by ataffanel