bitcraze/crazyflie-firmware #409: CF not responding to high-level commands after commit b2f01ff

tugayalperen, 2019-03-22:
Hi all,
In our setup we use Optitrack and crazyflie_ros package to make them fly. Up to commit number 624dcfb everything was working fine. After last commit, cf does not responds to any command like takeoff, go_to etc. There is no error while making, uploading or on the ROS side. When every other thing held the same, 624dcfb is working just fine but commit b2f01ff results with no response on cf. Probably something is going wrong with external position feed. Do you have any idea?

krichardsson, 2019-03-22:
@whoenig related to #408?

whoenig, 2019-03-22:
I did test with the Crazyswarm (which uses the high-level mode), but crazyflie_ros has a slightly different firmware code path (crazyswarm: `EXT_POSITION_PACKED` 
; crazyflie_ros `EXT_POSITION`). However, I am surprised about the observed behavior: if the position update does not succeed, the CF should still takeoff (and crash), but not simply have no response.

@tugayalperen: Which commit for crazyflie_ros did you use? Did you observe any crash in the firmware (fast blinking red LED)?

tugayalperen, 2019-03-25:
@whoenig: I am using th latest commit d498a18 (on 5 Feb) of crazyflie_ros. If you are asking about Led1, here in the video you can find it with latest firmware.

https://share.icloud.com/photos/0Qh2DO-p3gFPweJJboyRh3OJA

 Yet, I guess it is a normal heartbeat blinking. Because with the older firmware led is the same and everything works fine. I have tested the behavior again, it is again as I said, everything looks okay, I can even see Takeoff requested, GoTo requested etc. on the terminal but no response on cf, complete silence :)

whoenig, 2019-03-25:
Thanks for the detailed report. I found the issue and fixed it in a PR (see https://github.com/bitcraze/crazyflie-firmware/pull/415). Note that in my testing the CF did takeoff with the faulty firmware, but wouldn't be able to hold the correct position. So you might have some other issues in your setup. Most cases I have seen where the CF doesn't take off is when the IMU wasn't calibrated (your LED blink pattern looked fine; but I also added another PR [https://github.com/bitcraze/crazyflie-firmware/pull/413] to detect such cases more easily).

whoenig, 2019-03-28:
The fix has been merged. @tugayalperen, please confirm that your issue is resolved with this change so that we can close this issue.

tugayalperen, 2019-03-29:
@whoenig : When repo is cloned as instructed, there is on error respond to make command such as:

src/deck/drivers/src/usddeck.c: In function 'f_gets_without_comments':
src/deck/drivers/src/usddeck.c:238:5: error: array subscript has type 'char' [-Werror=char-subscripts]
     if (isspace(c)) {
     ^
cc1: all warnings being treated as errors
tools/make/targets.mk:26: recipe for target 'usddeck.o' failed
make[1]: *** [usddeck.o] Error 1
Makefile:327: recipe for target 'build' failed
make: *** [build] Error 2

It has nothing about this issue, this is for micro sd deck I guess, since I am not using this deck, by commenting out problematic section, I am able to make the project and fly.

whoenig, 2019-03-29:
I think we can close this issue then. I created a new issue for the build error you encountered (#418).

closed 2019-06-04 by krichardsson