bitcraze/crazyflie-firmware #418: build error in usddeck.c (array subscript has type)

whoenig, 2019-03-29:
@whoenig : When repo is cloned as instructed, there is on error respond to make command such as:

src/deck/drivers/src/usddeck.c: In function 'f_gets_without_comments':
src/deck/drivers/src/usddeck.c:238:5: error: array subscript has type 'char' [-Werror=char-subscripts]
     if (isspace(c)) {
     ^
cc1: all warnings being treated as errors
tools/make/targets.mk:26: recipe for target 'usddeck.o' failed
make[1]: *** [usddeck.o] Error 1
Makefile:327: recipe for target 'build' failed
make: *** [build] Error 2

It has nothing about this issue, this is for micro sd deck I guess, since I am not using this deck, by commenting out problematic section, I am able to make the project and fly.

_Originally posted by @tugayalperen in https://github.com/bitcraze/crazyflie-firmware/issues/409#issuecomment-477983043_

whoenig, 2019-03-29:
@tugayalperen: This was not picked up by the automated build tests. It is related to a PR I recently created. Could you share details about your build environment (which compiler? Do you have any outstanding changes? Any custom configuration in config.mk?).

BIsichei, 2019-05-20:
@whoenig I'm having this same error when trying to make under the "crazyflie-firmware" directory from the terminal. I cloned the [repo](https://github.com/bitcraze/crazyflie-firmware), initialized and updated submodules, and ran make. As far as I know i don't have any custom configurations. If it helps, i'm running this on laptop running ubuntu 16.04.

whoenig, 2019-05-21:
Updating your compiler should resolve the issue. I am running locally the following version:

```
$arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
```

tugayalperen, 2019-05-21:
I can confirm that, version change solves the problem.

BIsichei, 2019-05-21:
That fixed it, thanks

krichardsson, 2019-05-27:
Seems as this is solved? If not please re-open

closed 2019-05-27 by krichardsson

jungr-ait, 2019-06-24:
Hi,

I found a simple solution to that issue (https://github.com/USC-ACTLab/crazyflie-firmware/pull/1), just a simple C type cast, which has the nice side effect, that not everyone needs to upgrade the official compiler in Ubunutu 16.04. As @whoenig  said, we could reopen this issue and apply this change, if desired.

commit 0f71a82a (2019-07-02): Merge pull request #441 from jungr-ait/master

Fixes #418: compilation error due to isspace() in usddeck.c
commit 6f52ad2b (2019-07-02): Fixes #418: compilation error due to isspace() in usddeck.c

HINT: On Ubuntu 16.04 it resolves the error: array subscript has type
'char' [-Werror=char-subscripts] as all warnings are treated as errors.